tphakala/birdnet-go #453: Detection summary numbers dance when shrinking browser window

vibragiel, 2025-02-14:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Media queries are messing up the detection slot counts when the smaller viewport sizes are activated. The counts are incremented and moved around, event to future hours. It's better to see the attached video.

## Steps to Reproduce (for bugs)
1. Resize the browser window until the smaller viewports kick in.
2. See the cells with detection counts change numbers and positions.

## Expected behavior
The numbers should remain consistent and not move to future hours without detections.

## Actual behavior
The numbers change and move cells.

## Screenshots
https://github.com/user-attachments/assets/3703880c-c0e0-4ca4-bbe7-e5083899e54b

## Environment
### Version Information (build date)
Provide the following line from your system:
```
🐦 BirdNET-Go vunknown (built: 2025-02-10T15:37:02Z), using config file: /root/.config/birdnet-go/config.yaml
```

### System Information
- Device: 
  - [X] Raspberry Pi 5 Model B Rev 1.0
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: Raspberry Pi OS
- OS Version: Raspberry Pi OS bookworm
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: 8GB
  - Storage type and size: 256GB SSD

mathieu-mp, 2025-03-06:
Same here. Happen when I flip my phone, numbers get summed up, even sunrise and sunset logos appear twice each.

Whatever... Wife and I love this app sooo much, thanks a million times!

tphakala, 2025-03-06:
Yeah it is bug in responsive layout, I will fix it.

closed 2025-03-08 by tphakala