tphakala/birdnet-go #472: Always Exclude Species - Not working

Lotwook, 2025-02-18:
## Issue Type
- [X ] Bug report

## Description
In the Always Exclude Species I have excluded 3 species I would like excluded (Gray Wold, Engine, Siren) but they continue to be detected and continue to be sent to birdweather. I have attempted restarts after saving but results are the same.

## Steps to Reproduce (for bugs)
1. Go to settings > Species
2. In the Always Exclude section add species and save changes
3. Review dashboard or logs and see that detected excluded species are not dropped and recorded.

## Expected behavior
The detection should drop and now show in the dashboard or be sent to Birdweather

## Actual behavior
Detections continue to be logged and visible.

## Screenshots

[image]
[image]

## Environment
### Version Information (build date)
----------------------------------------------------------
 Add-on: Birdnet-go
 Realtime BirdNET soundscape analyzer
-----------------------------------------------------------
 Add-on version: 20250126
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.2
 Architecture: aarch64 / raspberrypi4-64
 Home Assistant Core: 2025.2.3
 Home Assistant Supervisor: 2025.02.1

### System Information
- Device: 
  - [X] Raspberry Pi (Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [X] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
- perhaps later I will
```

## Additional context
None

## Possible Solution
I don't

tphakala, 2025-02-18:
Please note that I am not author of Home Assitant add-on of BirdNET-Go and I cannot help very well with that version, and I see BirdNET-Go version you are using is already outdated.

Hoever it seems that range filter is not working at all for your deployment, have you set proper coordinates main settings section?

Lotwook, 2025-02-18:
My apologies I did post to the wrong github (doh!). I can bring it to the correct forum.  

To answer your question good spot! for sure it was set but somehow got reset (hmm).

closed 2025-02-18 by Lotwook