tphakala/birdnet-go #476: latest tag does not start successfully

Cassock, 2025-02-18:
## Issue Type
- [X] Bug report

## Description
Container ran fine for a few days, but as of a day or two ago, it does not start successfully using the compose below including `:latest` tag. Crashes with error `exec /usr/bin/birdnet-go: exec format error` in the container logs, and docker error message `The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64/v3) and no specific platform was requested   `. 

I was not around to know for sure, but I believe this coincided with a watchtower-induced upgrade

## Steps to Reproduce (for bugs)
```
  birdnet:
    image: ghcr.io/tphakala/birdnet-go:latest
    container_name: birdnet
    environment: 
      - ALSA_CARD=0
      - TZ=America/New_York
    volumes:
      - /docker/BirdNET/config:/config
      - /docker/BirdNET/data:/data
    ports:
      - 8080:8080
    devices:
      - /dev/snd
```

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
Docker host runs Ubuntu 22.04.3 LTS
- Additional hardware (if applicable):
  - USB Audio device: none

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [X] Yes, dev tag starts up with no immediate issues

tphakala, 2025-02-18:
Please pull latest image again, there was issue where latest image contained only linux/arm64 version

Cassock, 2025-02-18:
Back in biz. Thanks!

closed 2025-02-18 by Cassock