tphakala/birdnet-go #477: Settings changes in WebUI not saving

erdie721, 2025-02-19:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Whenever I save changes to any settings in the webUI it will not show changes on webpage reload. I can save config changes manually via editing of the configuration.yaml file. Not sure if it's related but Firefox also prompts me to save a password on each setting page even though I don't see any passwords that need saving. 

## Steps to Reproduce (for bugs)
1. Change any setting
2. Save changes
3. reload page

## Expected behavior
Saved settings take effect on page reload

## Actual behavior
No changes made on settings pages take effect

## Environment
### Version Information (build date)
i used the installation instructions from the documentation
Provide the following line from your system:
```
$ ./birdnet-go
no such file or directory
```

### System Information
- Device: 
  - [X] Raspberry Pi 4 Model B
  - OS: Raspberry Pi OS
- OS Version: 
Operating System: Debian GNU/Linux 12 (bookworm)  
Kernel: Linux 6.6.74+rpt-rpi-v8
Architecture: arm64

- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: 128GB microSD card
- Additional hardware (if applicable):
  - USB Audio device: UGreen USB adapter
  - Other peripherals: [e.g. Specific microphone, GPS module]

Hannsre, 2025-02-20:
Hi, 
I am facing the same issue using the Docker build. I can edit the `config.yml`, restart the stack and It'll work, but I can't set anything in the WebUI. 
Only error I see in the Logs is about not finding a hardware audio device, which is correct since I only use RTSP Streams for detection. 
I've tried pulling a new image, but looks like I am on the most recent version. There is no Version shown in the WebUI, just `unknown`. 

# System Information
##Device:

Intel NUC 10 
OS: Ubuntu 20.04 LTS
OS Version:
Operating System: Ubuntu 20.04 LTS
Kernel: Linux 5.4.0-174-generic x86_64
Architecture: amd64

##Hardware specs:

CPU: Intel Core i7-10710
RAM: 16GB
Storage type and size: 480GB Intel DC S4510 SSD

Hannsre, 2025-02-20:
Thanks, going with dev seems to have worked. 

There's still an issue with the site not refreshing after saving. So it looks like it's not saved, but it is. I.e: the `changed` info on top of the site remains after saving and it takes a manual reload to see whether the changes have been saved.

closed 2025-02-23 by tphakala