tphakala/birdnet-go #478: Not uploading data to Birdweather

pack2010, 2025-02-19:
Application has been working perfectly but then over last 12 hours am now getting this error message when trying to upload to Birdweather.

2025/02/19 08:12:59 Failed to decode JSON response: json: cannot unmarshal number into Go struct field .soundscape.duration of type string
2025/02/19 08:12:59 Failed to upload soundscape to Birdweather: json: cannot unmarshal number into Go struct field .soundscape.duration of type string
2025/02/19 08:12:59 error uploading to BirdWeather: json: cannot unmarshal number into Go struct field .soundscape.duration of type string
2025/02/19 08:12:59 Error executing action: json: cannot unmarshal number into Go struct field .soundscape.duration of type string

Is working fine to upload to the dashboard though.

Thanks!

sclitheroe, 2025-02-19:
I have the same issue - running the latest DEV tagged docker image.

tphakala, 2025-02-19:
This looks like a change on BirdWeather.com side, I'll provide a fix shortly.

tphakala, 2025-02-19:
I believe this is now fixed by https://github.com/tphakala/birdnet-go/pull/479, new dev docker image should be ready in 15 minutes or so

manicholls, 2025-02-19:
I pulled the latest dev down - 🐦 BirdNET-Go vunknown (built: 2025-02-19T14:53:19Z) but I'm still seeing the same failures around soundscape.duration.

tphakala, 2025-02-19:
Yeah it seems that docker images don't have version string set correctly, I need to fix that at some point. @manicholls can you please provide error logs you are seeing?

tphakala, 2025-02-19:
@manicholls yes you are correct, type should be float, I set it to int based on BirdWeather API documentation .. I'll fix this.

tphakala, 2025-02-21:
I tested latest fix and it seems to work at least for my nodes, I just released v0.6.2 with this fix included so I am closing this issue.

closed 2025-02-21 by tphakala