tphakala/birdnet-go #497: Settings not saving and excluded species not getting excluded

tescher, 2025-02-25:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Docker version installed on Ubuntu Intel NUC. Detections working great, dashboard shows fine. Tried to add excluded species (Engine and Dog), but the setting screen just hangs after clicking Save. Went into the Docker container as a bash shell and added them to config.yaml. They then showed up in settings, but still weren't excluded (even after restarting).

Also tried changing the MQTT login and password in settings, and that wouldn't save either. Just hangs after hitting Save.

## Steps to Reproduce (for bugs)
See above

## Expected behavior
[What you expected to happen]

## Actual behavior
[What actually happened]

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go

BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```
Not sure how to get this since I'm running the Docker version. Latest?

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [X] Other SBC Intel NUC
  - [ ] PC/Server (please specify)
- OS: Ubuntu
- OS Version: Ubuntu 22.04.5 LTS
- Hardware specs:
  - CPU:  Intel(R) Core(TM) i5-8259U CPU @ 2.30GHz
  - RAM: 32GB
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: Ugreen USB
  - Other peripherals: 500GB SSD

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [X] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```
I tried it, and it seemed to save the changes, but it was still detecting engines, and when I restarted the container, all settings changes (and detections) were gone.

## Additional context

## Possible Solution
?

joaldes, 2025-02-26:
I've had similar problems with the form not saving.  I was reviewing a detection and trying to save a false positive, which did nothing after I clicked save.  Same with the config settings; if I add directly to the yaml, it shows up.  Otherwise, nothing in the forms save.

 I checked my logs and have this about CSRF errors:

```
2025/02/25 20:34:33 [birdnet] 0.00 Bonasa umbellus_Ruffed Grouse_rufgro

2025/02/25 20:34:34 🚨 CSRF ERROR: Rejected request

2025/02/25 20:34:34 🔍 Request Method: POST, Path: /detections/review

2025/02/25 20:34:34 📌 CSRF Token in Header: QXWZpATJGeLBTdizPYPNwJlMqmxedBui

2025/02/25 20:34:34 📌 CSRF Token in Form: QXWZpATJGeLBTdizPYPNwJlMqmxedBui

2025/02/25 20:34:34 ⚠️ No CSRF Cookie found

2025/02/25 20:34:34 📝 All Cookies: _pk_id.1.72c8=2436bbf0e12f9076.1740540864.; _pk_ref.1.72c8=%5B%22%22%2C%22%22%2C1740540864%2C%22https%3A%2F%2F192.168.0.151%3A8006%2F%22%5D

2025/02/25 20:34:34 💡 Error Details: code=403, message=invalid csrf token

2025/02/25 20:34:34 [birdnet] results:

2025/02/25 20:34:34 [birdnet] 0.00 Human non-vocal_Human non-vocal_humnov
```

tphakala, 2025-03-08:
Sorry about delayed response. This is caused by CSRF (security feature) which was introduced recently, I have made several fixes related to these issues and I just released v0.6.3 version. Could you update to that and see if those issues are fixed for you?

https://github.com/tphakala/birdnet-go/releases/tag/v0.6.3

tescher, 2025-03-08:
Thanks. Is there a way to update the app in my Docker container without losing all my detection data?

tphakala, 2025-03-08:
All data, including config, is stored outside of container, so docker image update will not have any impact on detections.

tescher, 2025-03-08:
Hmm, I'm only seeing the .db files in the /data directory if I am in a shell in the container: 

docker exec -it 954d7275d5d2 /bin/bash

tphakala, 2025-03-08:
What is the command you use to start container with?

tescher, 2025-03-08:
docker run -ti \
  -p 8080:8080 \
  --env ALSA_CARD=1 \
  --env TZ="America/Chicago" \
  --device /dev/snd \
  -v /config \
  -v /data \
  ghcr.io/tphakala/birdnet-go:latest

tphakala, 2025-03-08:
and what is stored on your container host system's /data?

tescher, 2025-03-08:
Nothing in the host system /data. 

In the container's data is birdnet.db  birdnet.db-shm  birdnet.db-wal  clips

tphakala, 2025-03-08:
Huh. Then your container deployment has stored all data within container, data is likely in /root users home directory within container. Copy it out and please use install.sh to install new image

```
curl -fsSL https://github.com/tphakala/birdnet-go/raw/main/install.sh -o install.sh
bash ./install.sh
```

This is install method I can support as I know how it works. Data will be stored under /home/<user>/birdnet-go-app/

tescher, 2025-03-08:
OK I copied the data out, reinstalled, now I need to copy the data back into the host system /data and /config, but I can't get the docker container to remain stopped in order to do that. It keeps restarting itself. I don't want to copy the db files while it is running

tescher, 2025-03-08:
I think I got it after modifying the systemd service. Looks good.

Do you have a "buy me a coffee" account or something I can contribute to? This app is great.

closed 2025-03-16 by tphakala