tphakala/birdnet-go #501: install.sh doesnt use sodo for package install in the second run

Smo-RBR, 2025-02-27:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
🔧 Installing missing packages:  netcat-openbsd
E: Unable to locate package  netcat-openbsd
❌ Failed to install some packages
birdnet@birdnet:~ $ sudo apt install netcat-openbsd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libcamera0.1 libcamera0.2 linux-headers-6.1.0-18-arm64 linux-headers-6.1.0-18-common
  linux-headers-6.1.0-rpi7-common-rpi
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  netcat-openbsd

## Steps to Reproduce (for bugs)
Give install.sh the first run
log out/in
run install.sh a second time for continuing install

## Expected behavior
missing packages should be installed

## Actual behavior
because running apt without sudo, missing packages could not be installed

## Screenshots
pasted console output above

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```
still isnt installed

### System Information
- Device: 
  - [x] Raspberry Pi 3 (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian] Raspberry OS
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS] Bookworm

## Possible Solution
Add sudo to apt calls

tphakala, 2025-02-27:
install.sh does run apt with sudo.

can you please do manually

```
sudo apt update
sudo apt install netcat-openbsd
```
Does this install required netcat-openbsd package?

Smo-RBR, 2025-02-27:
Yes, it did this, and the installation worked flawless after manual installation. 

btw. respect for the installer, besides this little issue, its ultra comfortable.

tphakala, 2025-02-27:
Okay thanks for confirming, I need to make apt installations more robust, this is second issue ticket about same problem.

And I appreciate the feedback!

closed 2025-03-16 by tphakala