tphakala/birdnet-go #502: container keeps growing - clips now being deleted

uSlackr, 2025-02-28:
## Issue Type
- [x ] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
My container keeps growing - now at 10GB and disk space is 97.5%.  It appears clips aren't going away.  I found the following in the log:
`Error cleaning up clips:  parsing time "20250130T184446Z.mp3": extra text: ".mp3"`

My Retention Policy is `Usage`
Max Usage: `80%`
Minimum Clips: `10`

## Steps to Reproduce (for bugs)
1. Run container
2. Set Retention to Usage
3. Let run
4. Error logged at failed deletion

## Expected behavior
Disk space usage shouldn't grow past 80%
## Actual behavior
Disk fills

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2025-02-28:
Thank you for reporting, safety check for file deletions was not updated to support other than WAV files, this is now fixed. Fix is included in latest dev container image but I would recommend to wait for next nightly release if possible.

Also btw, if you don't have some specific need to use mp3 format I would commend agaist using it, it is very old and inefficient codec these days, AAC would be my recommendation.

uSlackr, 2025-02-28:
Thanks for this and the software.  I will await the next build.  One thing - is `audio export` the right name for this setting?  Perhaps `Audio Clip Save` would be clearer to the new user.
Thanks again!

uSlackr, 2025-04-07:
This appears to be back.  My container just went to 97% and retention is set to  80%.  I did a manual removal.  
I'm running on dev and just pulled over the weekend, but disk was over 80% on the prior dev build as well.

tphakala, 2025-04-07:
What does this show?

``` bash
journalctl -u birdnet-go.service -n 2000 | grep -i cleanup
```

uSlackr, 2025-04-07:
Nothing is returned.  I went all the way to 10,000

\\Greg

What does this show?

What does this show?

tphakala, 2025-04-08:
Was your deployment installed using install.sh or some other method? Are you seeing any logs if you run ```journalctl -u birdnet-go.service```

uSlackr, 2025-04-08:
I'm pretty sure I use install.sh

logs:  Looks like birdnet.service isn't running

Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 1.
Apr 08 09:51:13 birdnetlxc systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 08 09:51:13 birdnetlxc systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 09:51:13 birdnetlxc docker[747]: docker: Error response from daemon: driver failed programming external connectivity on endpoi>
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 2.
Apr 08 09:51:13 birdnetlxc systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 08 09:51:13 birdnetlxc systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 09:51:13 birdnetlxc docker[755]: docker: Error response from daemon: driver failed programming external connectivity on endpoi>
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 09:51:13 birdnetlxc systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 3.
Apr 08 09:51:14 birdnetlxc systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 08 09:51:14 birdnetlxc systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 09:51:14 birdnetlxc docker[769]: docker: Error response from daemon: driver failed programming external connectivity on endpoi>
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 4.
Apr 08 09:51:14 birdnetlxc systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 08 09:51:14 birdnetlxc systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 09:51:14 birdnetlxc docker[778]: docker: Error response from daemon: driver failed programming external connectivity on endpoi>
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 5.
Apr 08 09:51:14 birdnetlxc systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Start request repeated too quickly.
Apr 08 09:51:14 birdnetlxc systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 09:51:14 birdnetlxc systemd[1]: Failed to start birdnet-go.service - BirdNET-Go.

\\Greg

uSlackr, 2025-04-11:
I reran install.sh and the logs look better.

uSlackr, 2025-04-24:
Sorry to keep beating this horse, but it still isn't working.  
recent cleanup logs with cleanup debug set to true:
```
Apr 24 17:12:02 birdnetlxc 2fd9b1f366cc[239]: 2025/04/24 17:12:02 🧹 Running clip cleanup task
Apr 24 17:12:02 birdnetlxc 2fd9b1f366cc[239]: 2025/04/24 17:12:02 Starting cleanup process. Base directory: clips/, Threshold: 90.0%
Apr 24 17:12:02 birdnetlxc 2fd9b1f366cc[239]: 2025/04/24 17:12:02 Found 0 protected clips
Apr 24 17:12:03 birdnetlxc 2fd9b1f366cc[239]: 2025/04/24 17:12:03 No eligible audio files found for cleanup in clips/
Apr 24 17:12:03 birdnetlxc 2fd9b1f366cc[239]: 2025/04/24 17:12:03 🧹 Usage-based cleanup completed successfully, clips removed: 0, current disk utilization: 92%
```
And this from the config:
```
            retention:
                debug: true
                policy: usage
                maxage: 30d
                maxusage: 90%
                minclips: 10
```

Lenbok, 2025-12-14:
The cleanup thing seems not to be working correctly for me either. The target is 80% but I'm currently at 84%:

```
$ journalctl -u birdnet-go.service -n 2000 | grep -i cleanup | grep -v stream
[...]
Dec 15 09:43:19 piwakawaka docker[1205315]: 2025/12/15 09:43:19 🧹 Running clip cleanup task
Dec 15 09:45:04 piwakawaka docker[1205315]: 2025/12/15 09:45:04 🧹 Usage-based cleanup completed successfully, clips removed: 1000, current disk utilization: 84%
Dec 15 09:58:19 piwakawaka docker[1205315]: 2025/12/15 09:58:19 🧹 Running clip cleanup task
Dec 15 10:13:19 piwakawaka docker[1205315]: 2025/12/15 10:13:19 🧹 Running clip cleanup task
```

So it look likes it did remove 1000 clips at one point, but a) that wasn't enough to get below 80%, and b) subsequent checks didn't remove any more.

tphakala, 2026-10-07:
#4519 (merged into main, in the next build) changes how retention cleanup handles a backlog, which may cover what is reported here. Cleanup used to delete at most 1000 clips per run and then wait the whole check interval, so a disk filling faster than that never caught up. It now deletes continuously in paced batches until usage is back under the target, and starts the next run about a minute later when work is left.

The original `parsing time ... extra text` error is from an older build and is not related to this change.

If you still see usage staying above the target after updating, please post the `cleanup run summary` lines from `diskmanager.log`: their `stop_reason`, `locked_skipped`, `min_clips_blocked` and `usage_*_pct` fields show why a run stopped.