tphakala/birdnet-go #505: Recording duration vs. uploaded file duration

rammytheone, 2025-02-28:
Hi,
Why is there a difference between the length of the recording and the length of the uploaded file?
I don't know, maybe I messed something up during the update.

```
🐦 ESC[37mBirdNET-Go v0.6.2 (built: 2025-02-21T08:22:30Z), using config file: /root/.config/birdnet-go/config.yamlESC[0m
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using configured 1 threads of available 2 CPUs
System details: linux debian 12.9 on unknown hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 1.5, sensitivity: 1, interval: 15
```

```log
2025/02/28 16:45:56 Discarding detection of kos from source malgo due to false positive, matched 1/2 times
2025/02/28 16:46:21 Discarding detection of bernikla kanadyjska from source malgo due to false positive, matched 1/2 times
2025/02/28 16:46:30 Approving detection of kos from source malgo, matched 3 times
2025/02/28 16:46:30 Saved audio clip to clips/2025/02/turdus_merula_83p_20250228T164621Z.wav
2025/02/28 16:46:30 Saved audio clip to 2025/02/turdus_merula_83p_20250228T164621Z.wav
2025/02/28 16:46:30 detection time 16:46:19, begin time 2025-02-28 16:46:14.673606184 +0100 CET m=+332325.159534227, end time 2025-02-28 16:46:30.744296832 +0100 CET m=+332341.230224880
2025/02/28 16:46:31 Response Body: {"success":true,"soundscape":{"id":2162338861,"stationId":9327,"timestamp":"2025-02-28T16:46:19.000+01:00","url":"https://media.birdweather.com/soundscapes/ff9ae86f667411e33d91e026d277f11f","filesize":577580,"extension":"wav","duration":6.016,"mode":"birdnetpi"}}
2025/02/28 16:46:31 Soundscape successfully posted to Birdweather
2025/02/28 16:46:31 Posting detection to Birdweather:  https://app.birdweather.com/api/v1/stations/Tx7XkSg67gkeiJ5MMXJdiUfe/detections
2025/02/28 16:46:31 JSON Payload: {"timestamp":"2025-02-28T16:46:19.000+0100","lat":50.5809,"lon":18.0757,"soundscapeId":"2162338861","soundscapeStartTime":"2025-02-28T16:46:19.000+0100","soundscapeEndTime":"2025-02-28T16:46:22.000+0100","commonName":"kos","scientificName":"Turdus merula","algorithm":"2p4","confidence":"0.83"}
2025/02/28 16:46:32 Uploaded 2025/02/turdus_merula_83p_20250228T164621Z.wav to Birdweather
```

Maybe this is good, but it looks a bit weird on birdweather.com
[image]

tphakala, 2025-03-17:
This seems to be a problem at BirdWeather side, I reviewed BirdNET-Go side code and it correctly encodes audio in 3 second WAV file befor submission, BirdWeather then decides to render it as 6 second clip.

timsterc, 2025-03-20:
We save the raw soundscape that was uploaded -- and when I look at it in Raven -- it's 6sec.  Same in Audacity.

[image]

rammytheone, 2025-03-21:
I don't know, maybe something is wrong here?

Commits on Feb 19, 2025
fix: Change soundscape duration type to float64
fix: Update soundscape duration type from string to integer

https://github.com/tphakala/birdnet-go/commits/3c19a1f8d6c950d1d29eb742837c0602b81eb97a/internal/birdweather/birdweather.go?browsing_rename_history=true&new_path=internal%2Fbirdweather%2Fbirdweather_client.go&original_branch=main&since=2025-02-19&until=2025-02-19

closed 2025-04-06 by tphakala

rammytheone, 2025-04-07:
Thank you very much. It works fine on version nightly-20250406-9-gd8891ea