tphakala/birdnet-go #517: Translated labels in Docker container missing

Janhouse, 2025-03-04:
It is unclear where we are supposed to put the translation label files. Also, in the GUI there is Latvian language option, but in the repo there is no labels_lv.txt file, even though it exists in birdnet-pi. Could you please clarify how it is supposed to be set up?

tphakala, 2025-03-04:
It is not possible to add new labels, all label files are embedded in birdnet-go binary, label files included in repository are here
https://github.com/tphakala/birdnet-go/tree/main/internal/birdnet/data/labels

It seems that Lativan label file is missing, this a error I will fix. Latvian label file included in birdnet-pi seems to be corrupted, label files for current model file have to have exactly 6522 lines and it seems to have one less

```bash
wc -l labels_lv.txt
    6521 labels_lv.txt
```

Once I have found which label is missing I'll upload fixed label file to repository.

tphakala, 2025-03-04:
Latvian label file is now included, please update to latest nightly image either with docker pull or re-running install.sh and choosing update.

Janhouse, 2025-03-04:
@tphakala thanks! I did start the nightly container, it shows some problems with the label format

```
Species string has an unexpected format: Leptotila verreauxi_Bealtpieres ūbele
```

[image]

tphakala, 2025-03-04:
oh I see, one more fix is required. I'll push new update in hour or so.

closed 2025-03-07 by Janhouse