tphakala/birdnet-go #524: Query about sound stream

paulzwarts, 2025-03-08:
I keep having intermittent issues where the recorded detection audio played back from the audio widget for a bird is strange sounding, almost like an echo. To me it sounds like a duplicate audio playback a few millisecond apart. The waveform image also appears to be blurry like it printed the audio twice as well. So it seems like the source audio is glitching out. Is this a network issue? I am using Realtek (various modern models) cameras which is wired PoE, so in essence there should not be network congestion but I would not rule that out. But for the life of me, I cannot find a camera glitching out using other connectivity to the cameras to observe. Every day there will be a number of these detections with a corrupted audio. Also, if the source is corrupted, then I assume the detection is corrupted also because its listening to bad audio. There have been many quiet questionable detections with corrupt echoy audio that could have sounded like something else. 

Final side note question on audio and detections. I have coyote in my area which is detected as a number of different birds consistently like Trumpeted Swan. I'm guessing this is a Birdnet  issue with the people from Cornell? Any plans that the manual confirm/reject detection process be used as model refinement sorta like Frigate (and its + version) and double-take?

Many thanks, love the software, good work.

tphakala, 2025-03-08:
If source audio is corrupted, then yes both analysis and capture will both be impacted. With UDP transport network could impact RSTP audio, but TCP would detect and retry to correct any transmission errors and would eventually just cut off audio completely if network is poor enough. So if you are using UDP transport for RTSP, try using TCP and see if it help. I also have noticed that some PoE Cameras (I am using Reolink myself) require daily reboot to behave nicely.

I don't think that local refiniment of BirdNET AI model is very feasible, for one I haven't implemented any code required for model training and second it is very compute intensive process. I have been training my own classifier (AI model) to enhance BirdNET model, I have lots of audio clips of my local birds and some canines like red fox, running training process with this audio set takes almost 30 minutes on my Intel i7 13700k system. But I have been thinking that I could setup a system where people could submit audio samples and those would be used to improve model over time. Don't know if this will ever happen though, due to time and cost issues.

However, if you could record those coyote howls in long audio clips (longer the better, 10 - 30 minutes, including silence/traffic noise etc. which will be used as negative match) and send them to me I can add them to my classifier to improve coyote detection. I will include this improved model in BirdNET-Go some time before summer.

tphakala, 2025-03-08:
I have been recroding audio using ffmpeg, following command can be used to capture audio from RTSP source as 1 hour long flac clips. This can be run in a `screen` session so you don't need to keep console open while it records.

```bash
ffmpeg -rtsp_transport tcp -i rtsp://username:password@camera_ip:port/stream_path -c:a flac -compression_level 8 -f segment -segment_time 3600 -strftime 1 "output_%Y-%m-%d_%H-%M-%S.flac"
```

 Here is guide on how to use screen, https://linuxize.com/post/how-to-use-linux-screen/

Correct/False flagging is just a cosmetic feature at the moment, but it can be used for various filters etc. later. I don't have concrete roadmap but maybe I should write one, maybe it would help onboard more developers to help.

closed 2025-03-08 by paulzwarts