tphakala/birdnet-go #534: Spectograms are missing and the clips won't play

mathieu-mp, 2025-03-10:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Neither clips nor spectrograms are stored under the corresponding folder, nor they are accessible from the WebUI. It has previously worked, but there is not working any more.

## Steps to Reproduce (for bugs)
1. Access the `Home > Recent Detections` section
2. The spectogram is missing, and the clip won't play

## Expected behavior
The spectogram to display and the clip to play

## Actual behavior
The spectrogram is missing, the clip doesn't play, and a 404 is received when trying to download the file.

## Screenshots
Missing clips:
[image]

Previous working clips, still available:
[image]

## Logs during a detection
```
2025/03/10 15:09:42 Total time for processing all notes: 3.91474ms
2025/03/10 15:09:59 Buffer wrapped during write, adjusting start time to 2025-03-10 15:08:59.38025061 +0100 CET m=+6301.176311519
2025/03/10 15:10:24 Discarding detection of mésange charbonnière from source rtsp://rtsp-server:8554/stream due to false positive, matched 1/2 times
2025/03/10 15:10:50 Retrieved 5 detections in 522.28µs
2025/03/10 15:10:50 Total time for processing all notes: 3.873038ms
2025/03/10 15:10:59 Buffer wrapped during write, adjusting start time to 2025-03-10 15:09:59.407983308 +0100 CET m=+6361.204044217
2025/03/10 15:11:07 Approving detection of mésange charbonnière from source rtsp://rtsp-server:8554/stream, matched 2 times
2025/03/10 15:11:07 Failed to read audio segment from buffer: no capture buffer found for source: rtsp://rtsp-server:8554
2025/03/10 15:11:07 Error executing action: no capture buffer found for source: rtsp://rtsp-server:8554
2025/03/10 15:11:42 Retrieved 5 detections in 1.633583ms
2025/03/10 15:11:42 Total time for processing all notes: 4.150864ms
2025/03/10 15:11:50 Debug: ✅ CSRF token found in context for settings page: /settings/detectionfilters
2025/03/10 15:11:59 Buffer wrapped during write, adjusting start time to 2025-03-10 15:10:59.398796282 +0100 CET m=+6421.194857192
2025/03/10 15:12:04 Debug: ✅ CSRF token found in context for settings page: /settings/main
```
I guess the log from 15:11:07 matches the detection from 15:10:55.

Config is set as below:
```
debug: false
main:
    name: Birds!
    timeas24h: true
    log:
        enabled: true
        path: birdnet.log
        rotation: daily
        maxsize: 1048576
        rotationday: Sunday
birdnet:
    debug: false
    sensitivity: 1
    threshold: 0.85
    overlap: 1.5
    longitude: REDACTED
    latitude: REDACTED
    threads: 0
    locale: fr
    rangefilter:
        debug: false
        model: latest
        threshold: 0.01
    modelpath: ""
    labelpath: ""
    usexnnpack: true
realtime:
    interval: 15
    processingtime: false
    audio:
        source: ""
        ffmpegpath: ffmpeg
        soxpath: sox
        export:
            debug: true
            enabled: true
            path: clips/
            type: flac
            bitrate: 192k
            retention:
                debug: false
                policy: age
                maxage: 4w
                maxusage: 80%
                minclips: 20
        equalizer:
            enabled: false
            filters:
                - type: HighPass
                  frequency: 100
                  q: 0
                  gain: 0
                  width: 0
                  passes: 0
                - type: LowPass
                  frequency: 15000
                  q: 0
                  gain: 0
                  width: 0
                  passes: 0
    dashboard:
        thumbnails:
            debug: false
            summary: true
            recent: true
        summarylimit: 30
    dynamicthreshold:
        enabled: true
        debug: false
        trigger: 0.92
        min: 0.75
        validhours: 24
    log:
        enabled: false
        path: birdnet.txt
    birdweather:
        enabled: false
        debug: false
        id: ""
        threshold: 0.8
        locationaccuracy: 500
    privacyfilter:
        debug: false
        enabled: true
        confidence: 0.4
    dogbarkfilter:
        debug: false
        enabled: true
        confidence: 0.5
        remember: 1
        species: []
    rtsp:
        transport: tcp
        urls:
            - rtsp://rtsp-server:8554/stream
    mqtt:
        enabled: false
        broker: tcp://localhost:1883
        topic: birdnet
        username: birdnet
        password: secret
    telemetry:
        enabled: false
        listen: 0.0.0.0:8090
    species:
        include: []
        exclude: []
        config: {}
    weather:
        provider: yrno
        pollinterval: 60
        debug: false
        openweather:
            enabled: false
            apikey: ""
            endpoint: https://api.openweathermap.org/data/2.5/weather
            units: metric
            language: en
webserver:
    debug: false
    enabled: true
    port: "8080"
    log:
        enabled: false
        path: webui.log
        rotation: daily
        maxsize: 1048576
        rotationday: "0"
security:
    debug: false
    host: REDACTED
    autotls: false
    redirecttohttps: false
    allowsubnetbypass:
        enabled: false
        subnet: ""
    basicauth:
        enabled: true
        password: REDACTED
        clientid: REDACTED
        clientsecret: REDACTED
        redirecturi: REDACTED
        authcodeexp: 10m0s
        accesstokenexp: 1h0m0s
    googleauth:
        enabled: false
        clientid: REDACTED
        clientsecret: REDACTED
        redirecturi: REDACTED
        userid: REDACTED
    githubauth:
        enabled: false
        clientid: ""
        clientsecret: ""
        redirecturi: REDACTED
        userid: ""
    sessionsecret: REDACTED
output:
    file: {}
    sqlite:
        enabled: true
        path: birdnet.db
    mysql:
        enabled: false
        username: birdnet
        password: secret
        database: birdnet
        host: localhost
        port: "3306"
```

`ls -lhat` shows only old content:
```
:/docker/birdnet/data/clips/2025/03# ls -lhat
total 107M
drwxr-xr-x 2 root root  32K Mar  8 16:56 .
-rw-r--r-- 1 root root  10K Mar  8 16:56 corvus_monedula_100p_20250308T073048Z_400px.png
-rw-r--r-- 1 root root 8.6K Mar  8 12:06 streptopelia_decaocto_83p_20250308T112300Z_400px.png <- OK
-rw-r--r-- 1 root root  18K Mar  8 12:06 fringilla_coelebs_76p_20250308T104232Z_400px.png
-rw-r--r-- 1 root root  23K Mar  8 12:06 fringilla_coelebs_96p_20250308T104211Z_400px.png
-rw-r--r-- 1 root root 214K Mar  8 11:23 streptopelia_decaocto_83p_20250308T112300Z.flac <- OK
-rw-r--r-- 1 root root 282K Mar  8 10:42 fringilla_coelebs_76p_20250308T104232Z.flac
-rw-r--r-- 1 root root 307K Mar  8 10:42 fringilla_coelebs_96p_20250308T104211Z.flac
-rw-r--r-- 1 root root  20K Mar  8 09:57 pica_pica_76p_20250308T084646Z_400px.png
(...)
```
The filenames followed by the ` <- OK` sign successfully display, as per screenshot.

```
/docker/birdnet/config# df -h /docker
Filesystem      Size  Used Avail Use% Mounted on
/dev/nvme0n1p2  1.8T  1.4T  314G  82% /srv/dev-disk-by-uuid-a222283a-c3f1-4ccf-874b-3a77c8327c12
```

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
🐦 BirdNET-Go v0.6.3 (built: 2025-03-08T09:35:32Z), using config file: /root/.config/birdnet-go/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server: x64, dockerized
- OS: Debian 
- OS Version: Bookworm
- Hardware specs:
  - CPU: x64
  - RAM: 16Go
  - Storage type and size: SSD
- Additional hardware (if applicable):
  - USB Audio device: None
  - Other peripherals: RTSP

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [X] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

## Additional context
* Listening to the clips won't play even via other links displayed on the WebUI, et not only from the Recent Detections section

## Possible Solution
Sorry, not enough knowledge to answer that.

tphakala, 2025-03-10:
How did you install?
Did you use this process described on readme

```
curl -fsSL https://github.com/tphakala/birdnet-go/raw/main/install.sh -o install.sh
bash ./install.sh
```

If not, please install using this process

mathieu-mp, 2025-03-10:
I use this docker-compose file:
```
services:
  rtsp_server:
    image: bluenviron/mediamtx:latest
    container_name: rtsp-server
    restart: unless-stopped
    environment:
      MTX_RTMP: no
      MTX_HLS: no
      MTX_WEBRTC: no
      MTX_SRT: no
      MTX_RTSP: yes
      MTX_RTSPTRANSPORTS: tcp
      MTX_PATHS_STREAM_RUNONINITRESTART: yes
    ports:
      - 8554:8554
      
  birdnet_go:
    image: ghcr.io/tphakala/birdnet-go:latest
    container_name: birdnet-go
    tty: true
    stdin_open: true
    command: realtime --rtsp rtsp://rtsp-server:8554/stream
    restart: unless-stopped
    volumes:
      - /docker/birdnet/config:/config
      - /docker/birdnet/data:/data
    environment:
      TZ: Europe/Paris
    ports:
      - 8006:8080
    depends_on:
      - rtsp_server
```

Is it not valid? It worked until recently.

mathieu-mp, 2025-03-10:
After investigation, it looks related to the upgrade to v0.6.3, as it just works when downgrading to v0.6.2

tphakala, 2025-03-10:
Could you provide developer mode console logs from browser on page this happens?
Chome/Edge F12 and select Console tab.

mathieu-mp, 2025-03-10:
Here it is:
```
spectrogram:1
           Failed to load resource: the server responded with a status of 404 ()
spectrogram:1
           Failed to load resource: the server responded with a status of 404 ()
spectrogram:1
           Failed to load resource: the server responded with a status of 404 ()
audio:1
           Failed to load resource: the server responded with a status of 404 ()
audio:1 
           Failed to load resource: the server responded with a status of 404 ()
audio:1 
           Failed to load resource: the server responded with a status of 404 ()
audio:1 
           Failed to load resource: the server responded with a status of 404 ()
audio:1 
           Failed to load resource: the server responded with a status of 404 ()
audio:1 
           Failed to load resource: the server responded with a status of 404 ()
audioplayer.js?v=v0.6.3:708 Uncaught (in promise) NotSupportedError: The element has no supported sources.
togglePlay @ audioplayer.js?v=v0.6.3:708

```

As mentioned in the issue, the clips are not stored in the filesystem, and the spectrograms are also not created.

tphakala, 2025-03-10:
Does it behave same if you change audio file format to WAV? Wav is stored directly from go code, other formats use ffmpeg for file encoding and sounds like ffmpeg might not be working.

mathieu-mp, 2025-03-10:
I changed the config but the sun is set now and birds may not sing a lot. I will come back to you.

mathieu-mp, 2025-03-11:
Same with wav: clips and spectrograms are not stored and not playable.

tphakala, 2025-03-11:
is there any errors from birdnet-go container in journald logs?

mathieu-mp, 2025-03-11:
```bash
root@313b64821df0:/data# journalctl
bash: journalctl: command not found
```

And not much here...
```bash
root@313b64821df0:/data# ls -lha /var/log/
total 140K
drwxr-xr-x 1 root root 4.0K Mar  8 10:34 .
drwxr-xr-x 1 root root 4.0K Feb 24 01:00 ..
-rw-r--r-- 1 root root 5.9K Mar  8 10:34 alternatives.log
drwxr-xr-x 1 root root 4.0K Mar  8 10:34 apt
-rw-rw---- 1 root utmp    0 Feb 24 01:00 btmp
-rw-r--r-- 1 root root 114K Mar  8 10:34 dpkg.log
-rw-r--r-- 1 root root    0 Feb 24 01:00 faillog
-rw-r--r-- 1 root root  605 Mar  8 10:34 fontconfig.log
-rw-rw-r-- 1 root utmp    0 Feb 24 01:00 lastlog
-rw-rw-r-- 1 root utmp    0 Feb 24 01:00 wtmp
```

Can you guide me on where and what to check?

PS: Container output is provided in the initial post.

tphakala, 2025-03-11:
Thanks, log you provided gives me enough context what is going wrong

```text
Error executing action: no capture buffer found for source: rtsp://rtsp-server:8554
```

I'll look into fixing this later today

joriws, 2025-03-12:
I am using WAV saving and same problem applies with docker & v0.6.3. rtsp-camera source.

webui.log shows:
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/audio-level 200 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/top-birds?date=2025-03-12&_csrf=jRmeYTgqvuXlVUBoclzcyfKtRLuVhixJ 200 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/detections/recent?numDetections=5&_csrf=jRmeYTgqvuXlVUBoclzcyfKtRLuVhixJ 200 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/spectrogram?clip=2025%2f03%2fturdus_merula_94p_20250312T185103Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/spectrogram?clip=2025%2f03%2fcorvus_monedula_47p_20250312T173712Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/spectrogram?clip=2025%2f03%2fcorvus_monedula_86p_20250312T174425Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/spectrogram?clip=2025%2f03%2fpica_pica_83p_20250312T180221Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/spectrogram?clip=2025%2f03%2fpica_pica_74p_20250312T173405Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fpica_pica_83p_20250312T180221Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fturdus_merula_94p_20250312T185103Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fcorvus_monedula_47p_20250312T173712Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fcorvus_monedula_86p_20250312T174425Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fpica_pica_74p_20250312T173405Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fcorvus_monedula_47p_20250312T173712Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fcorvus_monedula_86p_20250312T174425Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fpica_pica_83p_20250312T180221Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fturdus_merula_94p_20250312T185103Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /api/v1/media/audio?clip=2025%2f03%2fpica_pica_74p_20250312T173405Z.wav 404 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /assets/images/favicon-16x16.png 200 <nil>
[2025-03-12T21:19:43+02:00] [INFO] 192.168.9.18 GET /assets/images/apple-touch-icon.png 200 <nil>
[2025-03-12T21:20:48+02:00] [INFO] 192.168.9.18 GET /api/v1/audio-level 200 <nil>
[2025-03-12T21:21:31+02:00] [INFO] 192.168.9.18 GET /api/v1/audio-level 200 <nil>

tphakala, 2025-03-13:
This is now fixed in codebase, problem was caused by RTSP URL sanitization which removed login credentials and path from RTSP URL when source was pushed to struct containing detection data.. This in turn broke audio clip saving process as audio save process was unable to find buffer based on sanitized URL 🤦‍♂️

I will release 0.6.4 soon.

I need to start writing more unit tests so these bugs would be detected before rolling a release ..

closed 2025-03-13 by tphakala

tphakala, 2025-03-13:
Fix is now included in latest nightly build

mathieu-mp, 2025-03-13:
Issue solved in the nightly-20250313 build