tphakala/birdnet-go #545: Where to store different model and label files

AxelHo, 2025-03-14:
Hi,
i want to use an older model file.
But i don't know where to store it using the docker version?
The webgui will not save the path, i edit t in the mapped config.yaml inside my home directory
Can anyone help me out?

Regards
Axel

tphakala, 2025-03-14:
Assuming you have installed BirdNET-Go using install.sh as described on README.md then you can copy model and label files in birdnet-go-app/data directory

```bash
$ ls -la birdnet-go-app/data/
total 132932
drwxr-xr-x 3 thakala thakala     4096 Mar  9 21:23 .
drwxr-xr-x 4 thakala thakala     4096 Mar  7 19:15 ..
-rw-r--r-- 1 root    root    10530816 Mar 14 18:05 birdnet.db
-rw-r--r-- 1 root    root       32768 Mar 14 18:07 birdnet.db-shm
-rw-r--r-- 1 root    root     4293072 Mar 14 18:07 birdnet.db-wal
-rw-r--r-- 1 thakala thakala 60464416 Mar  9 15:51 BirdNET-Go_v1.4.10.tflite
-rw-r--r-- 1 thakala thakala 60464384 Mar  7 19:13 BirdNET-Go_v1.4.9.tflite
drwxr-xr-x 3 root    root        4096 Feb 19 13:16 clips
-rw-r--r-- 1 thakala thakala   311135 Mar  9 15:02 labels_fi.txt
```

Model and label paths are relative to data path so you would configure it like

[image]

AxelHo, 2025-03-14:
thanks, that's what i needed. I want to use the 1.4 model with 1133 birds. It is enough for me here in Europe and maybe i get my owls back.
So i guess my labels are not correct. In the orignal file the scientific name is missing and the 2.4 model labels does not match.

2025/03/14 20:27:19 ❌ Error processing data for source malgo: error predicting species: mismatched labels and predictions lengths: 6522 vs 1133

Is there a correct label file in the repo

Lenbok, 2025-08-05:
@tphakala I want to use a labels file with customized (indigenous) common names in my install (but still use the bundled V2.4 model). 

I started with the BirdNET V2.4 en labels file and altered the desired species names, and placed the file under `birdnet-go-app/data/`. 

I then put the label file name into the "Label Path" settings UI as you show in the screenshot above.  However, it doesn't persist this setting when I save - does it require the "Model Path" to also be specified in order to use custom labels?

I also tried with the new svelte ui settings, and it *does* persist the "Label Path" setting, but even after a restart it doesn't seem to be applying the new labels.