tphakala/birdnet-go #558: RTSP sometimes still doesn't reconnect

xconverge, 2025-03-17:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Every few days, my rtsp connection drops for one reason or another. 

I use go2rtc for my rtsp source and it says 

`12:25:30.640 PM	debug	[rtsp] handle error=read tcp xxx.xxx.xxx.xxx:8554->xxx.xxx.xxx.xxx: i/o timeout`

Often times, birdnet-go reconnects fine, other times...it doesn't

I see `"2025/03/16 12:25:46 ⚠️ No data received from RTSP source rtsp:/xxx.xxx.xxx.xxx:8554/mic for 60 seconds, triggering restart"` in the birdnet-go log but it never actually comes back.

If I then restart my birdnet-go container (without touching my rtsp source or mic), all is well again and it picks up the stream and resumes working

## Steps to Reproduce (for bugs)
Not sure

## Expected behavior
Reconnects

## Actual behavior
[What actually happened]
Doesn't reconnect

## Environment
### Version Information (build date)
```
BirdNET-Go nightly-20250315 (built: 2025-03-15T10:40:10Z)
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server (please specify)
- OS: Debian

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [X] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

closed 2025-03-18 by tphakala

xconverge, 2025-03-19:
Thanks @tphakala, I am running this now!

tphakala, 2025-03-19:
There are some race conditions still left, I will submit more fixes.

aav7fl, 2025-03-31:
@tphakala 

Which edge cases are left? 

I've loaded up one of the newer dev builds (https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go/384541020?tag=7c665571219bbf7ec4c5410d0791c69a35fea6fe) and I've noticed that if my cameras reboot, BirdNET-Go never recovers those streams (although it does identify that they are silent in the UI and sometimes tries to recover them).

Sometimes I get this log:

```bash
2025/03/31 09:24:27 ⚠️ No data received from RTSP source rtsp://REDACTED@192.168.1.55:554/endpoint for 60 seconds, triggering restart
```

But it doesn't actually reconnect the ffmpeg stream unless I restart the docker container.

I can make a new issue if you'd like, but it seemed identical to this one.

tphakala, 2025-04-02:
@aav7fl PR #566 is still much work in progress, it will overhaul whole audio capture logic and I still need to fix some issues and test it well before mering. But I am glad that it did fix issues you had.