tphakala/birdnet-go #559: Summary displaying incorrectly after latest docker update

Sp4rkR4t, 2025-03-17:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

## Steps to Reproduce (for bugs)
1. pull the latest dev branch docker container
2. restart the container 

## Expected behavior
Summary displays one row for each species detected

## Actual behavior
Summary is displaying one row for EVERY detection of a species

## Screenshots

[image]

## Environment
### Version Information (build date)
🐦 BirdNET-Go unknown (built: 2025-03-16T18:42:53Z), using config file: /root/.config/birdnet-go/config.yaml

### System Information
- Device: 
  - [X] Raspberry Pi 4b 4Gb
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: Raspberry Pi OS
- OS Version: [e.g. Raspberry Pi OS Bookworm
- Hardware specs:
  - CPU: Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz
  - RAM: 4GB
  - Storage type and size: 512GB SSD
- Additional hardware (if applicable):
  - USB Audio device: Ugreen usb soundcard
  - Other peripherals: LAV microphone
### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [X] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

jtstockton, 2025-03-17:
Seeing the same issue that Sp4rkR4t reports, on nightly-20250316-8-ga98cb68

[image]

tphakala, 2025-03-17:
Fixed in https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250317

Sp4rkR4t, 2025-03-17:
Confirmed. Thanks.

closed 2025-03-17 by Sp4rkR4t