tphakala/birdnet-go #567: 0.6.4 manifest unknown

rammytheone, 2025-03-19:
I can't download version 0.6.4

```
root@bachata:~/birdnet-go# vi compose.yml
root@bachata:~/birdnet-go# cat compose.yml
services:
  birdnet-go:
    restart: unless-stopped

    image:  ghcr.io/tphakala/birdnet-go:v0.6.4
    ports:
      - 80:8080
    environment:
      - ALSA_CARD=0
      - TZ=Europe/Warsaw
    devices:
      - /dev/snd:/dev/snd
    volumes:
      - $HOME/birdnet-go/config:/config
      - $HOME/birdnet-go/data:/data
root@bachata:~/birdnet-go# docker compose --verbose pull && docker compose up -d
TRAC[0000] Docker Desktop integration not enabled
[+] Pulling 1/1
 ✘ birdnet-go Error manifest unknown                                                                                                                                          0.9s
DEBU[0000] otel error                                    error="<nil>"
DEBU[0000] otel error                                    error="<nil>"
Error response from daemon: manifest unknown
```

tphakala, 2025-03-19:
Looks like tagging is messed up, again. Use latest tag instead, it points to v0.6.4 image.

rammytheone, 2025-03-19:
Yes, 'latest' works. Thanks. 
Except that it downloaded 'nightly-20250315 (built: 2025-03-15T10:40:10Z)'

tphakala, 2025-03-19:
That nightly-20250315 tag is just false on that build, it is built on v0.6.4 codebase. However I would recommend using nightly builds anyway.

closed 2025-04-02 by tphakala