tphakala/birdnet-go #568: Audio sourced no longer recognized after update

hannorein, 2025-03-19:
## Issue Type
- [X ] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
After an update to the latest build, birdnet-go no longer recognized the sound card ("no audio sources available"). 

## Steps to Reproduce (for bugs)
1. Install an older version of(maybe from ~4 weeks ago)
2. Run the install script and select "check for updates". 
3. Birdnet-go no longer sees any audio devices.
4. Restarting docker/reboot doesn't help.

## Expected behavior
Updated should not change audio devices.

## Actual behavior
Somehow audio devices got messed up.

## Environment
### Version Information (build date)
Latest nightly dev built

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ X ] PC/Server (please specify)
- OS: Debian
- OS Version: 6.12.9
- Additional hardware (if applicable):
  - USB Audio device: Two (onboard and USB), neither of them gets recognized in birdnet-go anymore

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ X ] Yes
- [ ] No

## Additional context
The configuration file still lists
```
realtime:
    audio:
        source: USB Audio Device, USB Audio
```

hannorein, 2025-03-19:
The string seems slightly different. But even if it's the wrong string in the config file, the settings menu should allow me to select a new audio source (it showed none).

```
$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC897 Analog [ALC897 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 2: ALC897 Alt Analog [ALC897 Alt Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

```
at /etc/systemd/system/birdnet-go.service
[Unit]
Description=BirdNET-Go
After=docker.service
Requires=docker.service

[Service]
Restart=always
ExecStart=/usr/bin/docker run --rm \
    -p 8080:8080 \
    --env TZ="US/Eastern" \
    --device /dev/snd \
    -v /home/rein/birdnet-go-app/config:/config \
    -v /home/rein/birdnet-go-app/data:/data \
    ghcr.io/tphakala/birdnet-go:nightly

[Install]
WantedBy=multi-user.target
```

tphakala, 2025-03-19:
Yes, it is true that web UI should show audio devices regardless. 

This is likely a problem with /dev/snd mapping to container, is user your are running container as a member of audio group? 
Could something have changed /dev/snd permissions etc?

hannorein, 2025-03-19:
No. Nothing has changed on the machine at all. And again, it works perfectly fine (showing all the audio devices) if I use the same built but select "new install" instead of "update".

tphakala, 2025-03-19:
I just updated one of my nodes

```bash
$ curl -fsSL https://github.com/tphakala/birdnet-go/raw/main/install.sh -o install.sh
bash ./install.sh
 ____  _         _ _   _ _____ _____    ____
| __ )(_)_ __ __| | \ | | ____|_   _|  / ___| ___
|  _ \| | '__/ _` |  \| |  _|   | |   | |  _ / _ \
| |_) | | | | (_| | |\  | |___  | |   | |_| | (_) |
|____/|_|_|  \__,_|_| \_|_____| |_|    \____|\___/

🐦 BirdNET-Go Installation Script
This script will install BirdNET-Go and its dependencies.
🔍 Found existing BirdNET-Go installation
1) Check for updates
2) Fresh installation
3) Uninstall BirdNET-Go
4) Exit
❓ Select an option (1-4): 1
🌐 Checking network connectivity...

📡 Testing HTTP/HTTPS connectivity...
✅ HTTPS connection successful to https://github.com (HTTP 200)
✅ HTTPS connection successful to https://raw.githubusercontent.com (HTTP 301)
✅ HTTPS connection successful to https://ghcr.io (HTTP 301)

📡 Testing GitHub registry connectivity...
✅ GitHub registry is accessible

✅ Network connectivity check passed

🔄 Checking for updates...
🛑 Stopping BirdNET-Go service...
📥 Pulling latest nightly image...
nightly: Pulling from tphakala/birdnet-go
d9b636547744: Pull complete
421bbf8f5196: Pull complete
067f27c4e378: Pull complete
4f4fb700ef54: Pull complete
79a7150cdb83: Pull complete
1ea4b6ad5f38: Pull complete
7eca68bfc61e: Pull complete
06c16c689bde: Pull complete
3a0adc824402: Pull complete
cd48f17ef84c: Pull complete
Digest: sha256:f80cfd196b24e815bb2f631b25eaa94bfd2ae9eaa159b6df24b25b1e1a4f7e87
Status: Downloaded newer image for ghcr.io/tphakala/birdnet-go:nightly
ghcr.io/tphakala/birdnet-go:nightly
🚀 Starting BirdNET-Go service...
✅ Update completed successfully
thakala@mokki-pi4:~ $ journalctl -t docker -f
Mar 19 14:59:16 mokki-pi4 docker[42859]: Logging disabled
Mar 19 14:59:16 mokki-pi4 docker[42859]: 2025/03/19 14:59:16 Job queue started with max capacity of 1000 jobs
Mar 19 14:59:16 mokki-pi4 docker[42859]: 2025/03/19 14:59:16 JSON API v2 initialized at /api/v2
Mar 19 14:59:16 mokki-pi4 docker[42859]: HTTP server started on port 8080 (AutoTLS: false)
Mar 19 14:59:16 mokki-pi4 docker[42859]: 2025/03/19 14:59:16 Clip retention policy: usage
Mar 19 14:59:16 mokki-pi4 docker[42859]: ⇨ http server started on [::]:8080
Mar 19 14:59:16 mokki-pi4 docker[42859]: Available Capture Sources:
Mar 19 14:59:16 mokki-pi4 docker[42859]:   0: Discard all samples (playback) or generate zero samples (capture), null
Mar 19 14:59:16 mokki-pi4 docker[42859]:   1: AMS-24, USB Audio, :1,0 (✅ selected)
Mar 19 14:59:16 mokki-pi4 docker[42859]: Listening on source: AMS-24, USB Audio (:1,0)
Mar 19 14:59:32 mokki-pi4 docker[42859]: 2025/03/19 14:59:32 Finished initializing BirdImageCache (42 species needed images)
```

Audio device is working fine.

I can't help without additional details.

hannorein, 2025-03-19:
Ok. Thanks for looking into it. I just wanted to let you know in case others have a similar issue. I'm just using a fresh install now and that works just fine.

closed 2025-03-19 by hannorein