tphakala/birdnet-go #585: Daily summary navigation buttons on default dashboard don't work until a date is selected

phobiac, 2025-04-01:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
The left and right arrows to decrement/increment what day is selected for the daily summary do nothing until a date is manually set by clicking the date field.

## Steps to Reproduce (for bugs)
1. Load the default dashboard
2. Try to go back one day to see what birds were detected previously
3. Notice you're unable to until you set the date

## Expected behavior
I would expect the daily summary to default to showing today's date (which it does) and the date picker to reflect that, so clicking the buttons will by default navigate away from today's date.

## Actual behavior
No UI change from clicking the navigation buttons.

## Possible Solution
Instead of defaulting to no date in the date picking field, default to today's date.

tphakala, 2025-04-02:
Lots of details are missing from issue report.
1. Which version of BirdNET-Go this is about?
2. Which browser you are using?
3. Which OS are you using?

I have never seen this behaviour myself so I need a lot more details to work on this.

phobiac, 2025-04-02:
I'm seeing this behavior from the Home Assistant addon, which per its changelog is using the latest March 15th release branch.

``
🐦 BirdNET-Go nightly-20250315 (built: 2025-03-19T09:13:54Z), using config file: /root/.config/birdnet-go/config.yaml
``

I am seeing it in both Firefox and Edge browsers.

Unfortunately I'm not able to easily pull in the dev branch into HA and don't want to set up an entirely second instance for a minor UI bug, but this is a screenshot of what I'm seeing. This is immediately after loading the dashboard, I've not interacted with the page at all. Note that the date field is empty but it is as expected showing me results from today. In this state the navigation buttons do nothing.

[image]

aav7fl, 2025-04-02:
@phobiac For what it's worth, I have an recent PR that _might_ indirectly fix your issue. It prevents the dashboard page from pushing the date into the URL (`/#2025-04-02`) when you land on the root path (`/`) and always loads today's date in the picker. 

If you change the date from the picker, then it _will_ push the date to the URL. But it sounds like that behavior worked for you anyway as long as the date picker was already filled in.

https://github.com/tphakala/birdnet-go/pull/575

closed 2025-04-04 by tphakala