tphakala/birdnet-go #609: Failed to start birdnet-go.service - BirdNET-Go.

zanzeoo, 2025-04-07:
Hello 

i have some issues installing BIRDNET GO on my PIE 5 

While the installation i have this message : Failed to start birdnet-go.service - BirdNET-Go.

And the page at mylocalip is a caddy page " Caddy Works"

Any idea what's wrong ?

thx

tphakala, 2025-04-07:
How did you do install?

BirdNET-Go does not use nor ship with Caddy, do you have something else installed on same system which might be confilicting with BirdNET-Go installation?

PaulOckenden, 2025-04-19:
I just tried to upgrade from two versions ago to the latest nightly. Service wouldn't start.

So I did a full re-install. It's still broken.

Docker install on a Pi 5.

```
$ journalctl -fu birdnet-go.service
Apr 19 11:37:42 pi-500 docker[5059]: Adding new user `birdnet' to supplemental / extra groups `users' ...
Apr 19 11:37:42 pi-500 docker[5059]: Adding user `birdnet' to group `users' ...
Apr 19 11:37:42 pi-500 docker[5059]: chown: cannot access '/data/*': No such file or directory
Apr 19 11:37:43 pi-500 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=1/FAILURE
Apr 19 11:37:43 pi-500 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 19 11:37:43 pi-500 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 5.
Apr 19 11:37:43 pi-500 systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 19 11:37:43 pi-500 systemd[1]: birdnet-go.service: Start request repeated too quickly.
Apr 19 11:37:43 pi-500 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 19 11:37:43 pi-500 systemd[1]: Failed to start birdnet-go.service - BirdNET-Go.

```

PaulOckenden, 2025-04-19:
More info:

```
sudo systemctl status birdnet-go
× birdnet-go.service - BirdNET-Go
     Loaded: loaded (/etc/systemd/system/birdnet-go.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2025-04-19 11:46:24 BST; 1min 57s ago
   Duration: 555ms
    Process: 3680 ExecStart=/usr/bin/docker run --rm --name birdnet-go -p 8080:8080 --env TZ=Europe/London --env BIRDNET_UID=1000 --env BIRDNET_GID=1000 --add-host=host.docker.internal:host-gateway --device /dev/snd -v /home/paulockenden/birdnet-go-app/config:/config -v /home/paulockenden/birdnet-go-app/data:/data ghcr.io/tphakala/birdnet-go:nightly (code=exited, status=1/FAILURE)
   Main PID: 3680 (code=exited, status=1/FAILURE)
        CPU: 27ms

Apr 19 11:46:24 pi-500 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 5.
Apr 19 11:46:24 pi-500 systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Apr 19 11:46:24 pi-500 systemd[1]: birdnet-go.service: Start request repeated too quickly.
Apr 19 11:46:24 pi-500 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 19 11:46:24 pi-500 systemd[1]: Failed to start birdnet-go.service - BirdNET-Go.
```

PaulOckenden, 2025-04-19:
Shout if there's any further debug you'd like me to run.

tphakala, 2025-04-19:
do 

```
mkdir birdnet-go-app/data/clips
sudo systemctl start birdnet-go
```

PaulOckenden, 2025-04-19:
> do
> 
> ```
> mkdir birdnet-go-app/data/clips
> sudo systemctl start birdnet-go
> ```
> 
> it is a bug I caused by changing docker image not to run birdnet-go as root, I will fix it soon

Yup. That fixed it. Thanks,

PaulOckenden, 2025-04-19:
Updated to the latest nightly and now get this:

Client sent an HTTP request to an HTTPS server.

Was previously working fine as plain old port 80 http.

PaulOckenden, 2025-04-19:
> Updated to the latest nightly and now get this:
> 
> Client sent an HTTP request to an HTTPS server.
> 
> Was previously working fine as plain old port 80 http.

A full reinstall sorted it.

tphakala, 2025-04-20:
> do
> 
> ```
> mkdir birdnet-go-app/data/clips
> sudo systemctl start birdnet-go
> ```
> 
> it is a bug I caused by changing docker image not to run birdnet-go as root, I will fix it soon

Fixed with https://github.com/tphakala/birdnet-go/commit/962c7d03a74d6da1571837493d3214bce1218bb4

> Hello
> 
> i have some issues installing BIRDNET GO on my PIE 5
> 
> While the installation i have this message : Failed to start birdnet-go.service - BirdNET-Go.
> 
> And the page at mylocalip is a caddy page " Caddy Works"
> 
> Any idea what's wrong ?
> 
> thx

No further replies to this so closing issue

closed 2025-04-20 by tphakala