tphakala/birdnet-go #614: Failed to start service after installation

JJNBCaster, 2025-04-08:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [X] Question / Support

## Description

🚀 Starting BirdNET-Go...
✅ BirdNET-Go service started successfully!

🐳 Waiting for container to start...
⏳ Waiting for container to start (attempt 1/30)...
⏳ Waiting for container to start (attempt 2/30)...
⏳ Waiting for container to start (attempt 3/30)...
❌ Service stopped unexpectedly
Checking service logs:
Apr 08 19:36:31 ip-172-31-28-222 systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 19:36:31 ip-172-31-28-222 docker[3199]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
Apr 08 19:36:31 ip-172-31-28-222 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
Apr 08 19:36:31 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:31 ip-172-31-28-222 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 1.
Apr 08 19:36:31 ip-172-31-28-222 systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 19:36:32 ip-172-31-28-222 docker[3277]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 2.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 19:36:32 ip-172-31-28-222 docker[3331]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 3.
Apr 08 19:36:32 ip-172-31-28-222 systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 19:36:33 ip-172-31-28-222 docker[3398]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 4.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: Started birdnet-go.service - BirdNET-Go.
Apr 08 19:36:33 ip-172-31-28-222 docker[3451]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 5.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Start request repeated too quickly.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Apr 08 19:36:33 ip-172-31-28-222 systemd[1]: Failed to start birdnet-go.service - BirdNET-Go.

## Steps to Reproduce (for bugs)
1. [First Step] Installed again to new instance
2. [Second Step] Installed again to new instance
3. [and so on...]

## Expected behavior
I have installed this 5+ times and always worked fine

## Actual behavior
Didnt start

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server (please specify)
24.04.2 LTS (GNU/Linux 6.8.0-1026-aws x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/pro

 System information as of Tue Apr  8 19:29:14 EEST 2025

  System load:  0.14              Temperature:           -273.1 C
  Usage of /:   8.6% of 28.02GB   Processes:             117
  Memory usage: 30%               Users logged in:       0
  Swap usage:   0%   

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [X] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

drkpxl, 2025-04-14:
Seeing the same issue on my Pi4

drkpxl, 2025-04-14:
```
Apr 14 10:04:27 birds systemd[1]: Starting docker.service - Docker Application Container Engine...
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.257241066-06:00" level=info msg="Starting up"
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.261716690-06:00" level=info msg="[core] parsed scheme: \"unix\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.261783907-06:00" level=info msg="[core] scheme \"unix\" not registered, fallback to default scheme" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.261845278-06:00" level=info msg="[core] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.261876861-06:00" level=info msg="[core] ClientConn switching balancer to \"pick_first\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.261905650-06:00" level=info msg="[core] Channel switches to new LB policy \"pick_first\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.262002878-06:00" level=info msg="[core] Subchannel Connectivity change to CONNECTING" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.262178110-06:00" level=info msg="[core] Subchannel picks a new address \"unix:///run/containerd/containerd.sock\" to connect" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.262869602-06:00" level=info msg="[core] Channel Connectivity change to CONNECTING" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.263094123-06:00" level=info msg="[core] Subchannel Connectivity change to READY" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.263148741-06:00" level=info msg="[core] Channel Connectivity change to READY" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265053684-06:00" level=info msg="[core] parsed scheme: \"unix\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265118201-06:00" level=info msg="[core] scheme \"unix\" not registered, fallback to default scheme" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265196113-06:00" level=info msg="[core] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265234227-06:00" level=info msg="[core] ClientConn switching balancer to \"pick_first\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265274320-06:00" level=info msg="[core] Channel switches to new LB policy \"pick_first\"" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265346163-06:00" level=info msg="[core] Subchannel Connectivity change to CONNECTING" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265457323-06:00" level=info msg="[core] Subchannel picks a new address \"unix:///run/containerd/containerd.sock\" to connect" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.265516289-06:00" level=info msg="[core] Channel Connectivity change to CONNECTING" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.266480870-06:00" level=info msg="[core] Subchannel Connectivity change to READY" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.266651051-06:00" level=info msg="[core] Channel Connectivity change to READY" module=grpc
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.984993292-06:00" level=warning msg="Unable to find memory controller"
Apr 14 10:04:27 birds dockerd[17624]: time="2025-04-14T10:04:27.986003333-06:00" level=info msg="Loading containers: start."
Apr 14 10:04:28 birds dockerd[17624]: time="2025-04-14T10:04:28.466858285-06:00" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferred IP address"
Apr 14 10:04:28 birds dockerd[17624]: time="2025-04-14T10:04:28.662671230-06:00" level=info msg="Loading containers: done."
Apr 14 10:04:28 birds dockerd[17624]: time="2025-04-14T10:04:28.848882716-06:00" level=info msg="Docker daemon" commit=5d6db84 graphdriver(s)=overlay2 version=20.10.24+dfsg1
Apr 14 10:04:28 birds dockerd[17624]: time="2025-04-14T10:04:28.849226135-06:00" level=info msg="Daemon has completed initialization"
Apr 14 10:04:28 birds systemd[1]: Started docker.service - Docker Application Container Engine.
Apr 14 10:04:28 birds dockerd[17624]: time="2025-04-14T10:04:28.934932460-06:00" level=info msg="API listen on /run/docker.sock"
Apr 14 10:12:28 birds dockerd[17624]: time="2025-04-14T10:12:28.257694152-06:00" level=info msg="ignoring event" container=e9d72b5239a302fca42b2f5a495c7adffb6d54473e32461b517f9aafc585e85f module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Apr 14 10:12:30 birds dockerd[17624]: time="2025-04-14T10:12:30.166307891-06:00" level=info msg="ignoring event" container=7ae899755d5d0c22115d455d712a2ba940eb9ccc8d3a2a80cd87f4bcc37caa67 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
[... cut ...] nnection is closing"
Apr 14 10:13:14 birds systemd[1]: Stopping docker.service - Docker Application Container Engine...
Apr 14 10:13:14 birds dockerd[17624]: time="2025-04-14T10:13:14.568325924-06:00" level=info msg="Daemon shutdown complete"
Apr 14 10:13:14 birds systemd[1]: docker.service: Deactivated successfully.
Apr 14 10:13:14 birds systemd[1]: Stopped docker.service - Docker Application Container Engine.
Apr 14 10:13:14 birds systemd[1]: docker.service: Consumed 1min 37.798s CPU time.
```

drkpxl, 2025-04-14:
If you change the local to uk it will work. there is no US local which is annoying

tphakala, 2025-04-14:
There is us locale, en-us

drkpxl, 2025-04-14:
Yes but you are not correctly setting it in your setup script, you are setting en, not en-us so its failing. And because your have -rm in your systemctl it's a pain to debug. :)

Changing it to en-us would work, but then again you aren't actually using the local config file either so you need to add that argument to systemctl as well.

I got it to work, but the easist fix would be A - updating your setup script to correctly apply en-us, and use the local config.yml by adding this to your systemctl --env CONFIG_PATH=/config/config.yaml

drkpxl, 2025-04-14:
https://github.com/tphakala/birdnet-go/pull/628 should fix it, I didn't didn't locally since I got it up running manually but pretty straightforward

tphakala, 2025-04-15:
install.sh is now fixed to match new locale codes with https://github.com/tphakala/birdnet-go/pull/629. Thank you @drkpxl for PR but more changes were required to fully implement all locales changes.

closed 2025-04-15 by tphakala