tphakala/birdnet-go #618: Not using correct local bird names

PaulOckenden, 2025-04-10:
Pretty simple issue really: Birdnet-Go (wrongly*) identified and displayed "Eurasian Thick-knee". But I'm in the UK and the correct name for this bird is Stone-curlew.

Why I look at my data on the Birdweather website, if I set the language (in the cogwheel settings) to be English (US) it shows the same as BirdNET-Go,  Eurasian Thick-knee. But if I change the Berdweather settings to English (UK) it corrects the bird name to Stone-curlew.

I'm running BirdNET-Go on a Pi where the locale is set to UK, and my location in the Birdnet settings is also in the UK.

So shouldn't it be doing the same as Birdweather and showing me the UK bird names?

Thanks.

tphakala, 2025-04-13:
This is because BirdNET-Go did not have British English labels, only American English. This PR fixes this https://github.com/tphakala/birdnet-go/pull/620

closed 2025-04-13 by tphakala

PaulOckenden, 2025-04-13:
Ah, superb. I’ll wait for that to appear in a release. Thank you.