tphakala/birdnet-go #639: Enable optional retain flag on MQTT publish

davidski, 2025-04-19:
## Issue Type
- [ ] Bug report
- [X] Feature request
- [ ] Question / Support

## Description
When the MQTT support was merged, a coderabbit suggestion (see https://github.com/tphakala/birdnet-go/pull/98#discussion_r1553265709) was to make the retain flag configurable rather than hardcoded to false. This change was not incorporated.

Retain support is very useful for integrating with Home Assistant, where the state of HASS MQTT sensors is refreshed on HASS restart. With the retain flag set to fault, Birdnet-derived MQTT sensors are 'unknown' at start. With retain set to true, Home Assistant can get the last known state of the sensor and process that.

This method of setting retain on MQTT messages is how platforms such as Zigbee2MQTT preserve state of sensors across Home Assistant restarts.

## Steps to Reproduce (for bugs)
N/A

## Expected behavior
N/A

## Actual behavior
Retain flag is always set to false.

## Screenshots
N/A

## Environment
### Version Information (build date)
N/A

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [X] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.

Tested with dev image from 4/18.

## Additional context
N/A

## Possible Solution
A `retain` boolean under the mqtt session of the config.yaml would be a great place to allow the user to specify whether this flag should be set. While I prefer this to default to true, so long as it's configurable, it's all good. 😄 

Thanks for the terrific project!

tphakala, 2025-04-20:
This is now in, retain defaults to false but setting is available in MQTT settings in web UI with note helpful note for HomeAssistant users

closed 2025-04-20 by tphakala

davidski, 2025-04-20:
Thanks a bunch for the quick implementation here! ❤ 

The info text is very hard to read when using a dark theme. It looks like this block is wrapped in a text-info-content span block, which is not present on the similar block under the weather integration. If I remove that span setting, the text looks to have the same style (under Firefox/MacOS) as the weather integration.

[image]

Happy to open a separate issue for this if that helps, @tphakala!