tphakala/birdnet-go #648: New taxonomy labels/locale break with current config

aav7fl, 2025-04-21:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

The new taxonomy labels/locale config break BirdNET-Go from starting using my existing config.

I believe this was due to changes introduced here: https://github.com/tphakala/birdnet-go/pull/620

If using 

```yaml
birdnet:
    locale: en
```

Birdnet won't start up anymore with these errors:

```bash
Command execution error: failed to initialize BirdNET: failed to load labels: failed to load label file: label file for locale 'en' not found. Available files: [BirdNET_GLOBAL_6K_V2.4_Labels_af.txt BirdNET_GLOBAL_6K_V2.4_Labels_ar.txt BirdNET_GLOBAL_6K_V2.4_Labels_bg.txt BirdNET_GLOBAL_6K_V2.4_Labels_ca.txt BirdNET_GLOBAL_6K_V2.4_Labels_cs.txt BirdNET_GLOBAL_6K_V2.4_Labels_da.txt BirdNET_GLOBAL_6K_V2.4_Labels_de.txt BirdNET_GLOBAL_6K_V2.4_Labels_el.txt BirdNET_GLOBAL_6K_V2.4_Labels_en_uk.txt BirdNET_GLOBAL_6K_V2.4_Labels_en_us.txt BirdNET_GLOBAL_6K_V2.4_Labels_es.txt BirdNET_GLOBAL_6K_V2.4_Labels_fi.txt BirdNET_GLOBAL_6K_V2.4_Labels_fr.txt BirdNET_GLOBAL_6K_V2.4_Labels_he.txt BirdNET_GLOBAL_6K_V2.4_Labels_hr.txt BirdNET_GLOBAL_6K_V2.4_Labels_hu.txt BirdNET_GLOBAL_6K_V2.4_Labels_id.txt BirdNET_GLOBAL_6K_V2.4_Labels_is.txt BirdNET_GLOBAL_6K_V2.4_Labels_it.txt BirdNET_GLOBAL_6K_V2.4_Labels_ja.txt BirdNET_GLOBAL_6K_V2.4_Labels_ko.txt BirdNET_GLOBAL_6K_V2.4_Labels_lt.txt BirdNET_GLOBAL_6K_V2.4_Labels_ml.txt BirdNET_GLOBAL_6K_V2.4_Labels_nl.txt BirdNET_GLOBAL_6K_V2.4_Labels_no.txt BirdNET_GLOBAL_6K_V2.4_Labels_pl.txt BirdNET_GLOBAL_6K_V2.4_Labels_pt_BR.txt BirdNET_GLOBAL_6K_V2.4_Labels_pt_PT.txt BirdNET_GLOBAL_6K_V2.4_Labels_ro.txt BirdNET_GLOBAL_6K_V2.4_Labels_ru.txt BirdNET_GLOBAL_6K_V2.4_Labels_sk.txt BirdNET_GLOBAL_6K_V2.4_Labels_sl.txt BirdNET_GLOBAL_6K_V2.4_Labels_sr.txt BirdNET_GLOBAL_6K_V2.4_Labels_sv.txt BirdNET_GLOBAL_6K_V2.4_Labels_th.txt BirdNET_GLOBAL_6K_V2.4_Labels_tr.txt BirdNET_GLOBAL_6K_V2.4_Labels_uk.txt BirdNET_GLOBAL_6K_V2.4_Labels_zh.txt]
```

Updating to 

```yaml
birdnet:
    locale: English (US)
```

allows is to work. However, anytime there is a settings change, BirdNET-Go writes back `en` to the config file. Which will prevent BirdNET-Go from starting up again the next time it restarts.

So it:
- Breaks with existing configs for users who may be upgrading
- BirdNET-Go writes back config that isn't considered valid (unless allowing `en` works again)

## Steps to Reproduce (for bugs)
1. Change config to this:
```yaml
birdnet:
    locale: en
```
2. Restart
3. Update any birdnet settings that touch the config file
4. Restart
5. BirdNET-Go won't start

## Expected behavior

BirdNET-Go starts.

## Actual behavior

BirdNET-Go didn't start.

## Environment
### Version Information (build date)

Build from `e4899ae4d9e592348e38a84bdaee89266845cb02`

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2025-04-21:
Set locale to en-us

aav7fl, 2025-04-21:
> Set locale to en-us

I just tried setting it as `en-us` in the `config.yml`. This _does_ allow BirdNET to start.

But when change any setting in BirdNET, it writes back `en` as my locale in the `config.yml`, overwriting my `en-us` (which breaks on the next restart).

tphakala, 2025-04-21:
OK, thanks for reporting. En locale should not be listed anymore at all, only en-uk and en-us are valid english labels. Locale list and default locale in case of missing locale should now be fixed in latest nightly snapshot.

closed 2025-04-21 by tphakala

aav7fl, 2025-04-21:
> OK, thanks for reporting. En locale should not be listed anymore at all, only en-uk and en-us are valid english labels. Locale list and default locale in case of missing locale should now be fixed in latest nightly snapshot.

Thanks!

Would you mind updating the wiki here with those new supported language codes (or have it link back to the source code)? I don't want the next person who finds it to get confused.

https://github.com/tphakala/birdnet-go/wiki/BirdNET%E2%80%90Go-Guide#supported-locales-for-species-labels

tphakala, 2025-04-21:
Done