tphakala/birdnet-go #655: Timestamps are GMT (I set America/New_York)

y8s, 2025-04-23:
## Issue Type
- [ ] Bug report

## Description
Using docker compose when setting TZ it appears to ignore the local time and uses GMT.

## Steps to Reproduce (for bugs)
1. make compose file (via dockge)
2. start it, look at watch, wait for birds to cheep
3. see timestamp is +4 hrs compared to my watch

## Expected behavior
timestamp to match my local tz

## Actual behavior
7pm chirps and cheeps logged as 23:00 chirps and cheeps. all historical bird sounds are shifted as well.

## Screenshots

[image]

## Environment
### Version Information (build date)
Provide the following line from your system:

compose file
```
services:
  birdnet-go:
    image: ghcr.io/tphakala/birdnet-go:nightly-20250407
    # is actually v0.6.4
    ports:
      - 8080:8080
    environment:
      - TZ="America/New_York"
    volumes:
      - ./config:/config
      - ./data:/data
    command: realtime --rtsp rtsp://user@pass@/<redacted ip>:554/cam
networks: {}
```

### System Information
- Device: 
  - [ ] old dell
- OS: proxmox 8.3.5
- OS Version: debian bookworm vm

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
It doesn't run. everything newer than this version has some `realtime is not in the path` error.

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```
I tried all nightlies from 20250407 to 20250421 but only 20250407 works.

## Additional context
none

## Possible Solution
none

y8s, 2025-04-23:
I pulled the quotes right off the installation wiki. The docker command line version still has quotes here:
https://github.com/tphakala/birdnet-go/wiki/installation.md#manual-docker-installation-advanced-linux-only

I removed the quotes and the timestamps are correct now. Useful piece of info for docker env variables.

Thanks

AKHwyJunkie, 2025-04-25:
I ran into this same issue.  The fix for me was to edit /etc/timezone, which was set to GMT.  For thoroughness, I also ran "timedatectl set-timezone [region/location]"  It's been a good bit of time since I fixed this, but If I recall right, I also had to reinstall the container after this.  I think part of the installation script somehow references the system's timezone and doesn't reflect what's set in the container.