tphakala/birdnet-go #657: UID/GID ownership issues

xconverge, 2025-04-23:
Starting with https://github.com/tphakala/birdnet-go/pull/636 just wanted to make you aware, I am fine to stick to v0.6.4 for now

I can't run the nightlies due to the UID/GID changes

I have kind of a weird setup related to UID/GID

```
    birdnet-go:
        container_name: birdnet-go
        image: ghcr.io/tphakala/birdnet-go:(latest/v0.6.4 is fine, newer nightlies are not)
        environment:
            - TZ=${TZ}
            #- BIRDNET_UID=1000
            #- BIRDNET_GID=1000
        volumes:
            - "${DOCKER_VOLUMES}/birdnet-go/config:/config"
            - "/mnt/files/birds:/data"
        ports:
            - 9099:8080
        restart: always
```

My `/data` folder is on a synology NAS with a nfs share that is mounted to my Debian system running docker

Maybe I could move the db to my ssd and only keep the .wav files on the NFS mount somehow?

but whether I use BIRDNET_UID/GID as the defaults or 1000/1000 or the values which I think would make sense for the NFS mount and how I have it setup, I get forbidden chown commands all over the place and the container never starts up successfully

I have tried 1024 and 100 as well which is what should be right.
```
/mnt/files/birds$ ls -lh
total 14M
-rwxrwxrwx 1 1024 users 9.4M Apr 22 22:26 birdnet.db
-rwxrwxrwx 1 1024 users  32K Apr 22 22:37 birdnet.db-shm
-rwxrwxrwx 1 1024 users 4.0M Apr 22 22:38 birdnet.db-wal
drwxrwxrwx 1 1024 users    8 Feb 12 07:04 clips
```

xconverge, 2025-04-24:
I am silly, I had 1026 instead of 1024 for the BIRDNET_UID. I set it correctly and got the container started

xconverge, 2025-04-26:
After this one https://github.com/tphakala/birdnet-go/pull/660 I am back to unable to use nightlies, I am totally fine staying on the commit I am on (095d8ec2b3a0a3cc0f5f43677aa655eb52b47678) but wanted to at least let you know!

```
Starting BirdNET-Go with UID:1024, GID:100
Starting BirdNET-Go as user birdnet (1024:100)
🐦 BirdNET-Go nightly-20250424-42-g58c0f16 (built: 2025-04-26T00:31:45Z), using config file: /home/birdnet/.config/birdnet-go/config.yaml
2025/04/25 17:44:37 ✅ HLS streaming directory initialized at: /home/birdnet/.config/birdnet-go/hls
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
2025/04/25 17:44:37 🧹 Cleaned up leftover HLS streaming files
⚠️ Using legacy range filter modelSystem details: linux debian 12.10 on unknown hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 1.5, sensitivity: 1, interval: 15
2025/04/25 17:44:37 WARN: Unsupported database type 'SQLite' for image_caches schema check. Assuming schema is correct.
2025/04/25 17:44:37 Using global image provider registry
2025/04/25 17:44:37 Registered WikiMedia image provider
2025/04/25 17:44:37 Attempting to register AviCommons provider...
2025/04/25 17:44:37 Successfully registered AviCommons image provider
2025/04/25 17:44:37 Using WikiMedia as the default image provider (auto mode)
2025/04/25 17:44:38 Starting background image cache warm-up...
2025/04/25 17:44:38 Pre-fetched 1 cached image records for provider 'wikimedia'
2025/04/25 17:44:38 Pre-fetched 0 cached image records for provider 'avicommons'
2025/04/25 17:44:38 Job queue started with max capacity of 1000 jobs
2025/04/25 17:44:38 Cache warm-up: 65 species require image fetching.
Logging disabled
2025/04/25 17:44:38 Resolved relative media export path "clips/" to absolute path "/usr/bin/clips"
2025/04/25 17:44:38 Failed to initialize API controller: failed to create media export directory "/usr/bin/clips": mkdir /usr/bin/clips: permission denied

```

tphakala, 2025-04-26:
Okay, thanks to letting me know

xconverge, 2025-04-28:
@tphakala looks good now (on nightly-20250428 or later), nice work!

`2025/04/27 19:00:09 Resolved relative media export path "clips/" to absolute path "/data/clips"`

closed 2025-04-28 by xconverge