tphakala/birdnet-go #667: Audio Device Validation Failed

geojoek, 2025-04-25:
## Issue Type
- [ X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Running standalone install on Linux Mint, when starting ./birdnet go realtime, get error "⚠️ Audio device validation failed: configured audio device 'WordForum USB    Mono' not found", and while the webGUI allows one to select the device and it persists in the settings, it shows the device as silent and does not detect audio from it. The system audio monitor shows the device is working normally and detecting audio.

## Steps to Reproduce (for bugs)
1. Launched ./birdnet-go standalone
2. Loaded WebGUI
3. Selected Audio Device in  Settings
4. Restarted ./birdnet-go from command line, still does not find audio device.

## Expected behavior
To find audio device and show audio coming from it.

## Actual behavior
Does not find audio device, but still shows it in settings. Shows audio device as silent while system audio monitor is detecting audio from device.

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
🐦 BirdNET-Go v0.6.4 (built: 2025-03-19T09:13:07Z), using config file: /home/joe/.config/birdnet-go/config.yaml
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using configured 4 threads of available 8 CPUs
System details: linux linuxmint 22.1 on unknown hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 2, sensitivity: 1, interval: 30
2025/04/25 15:15:04 BirdImageCache initialized (all species images already cached)
Logging disabled
2025/04/25 15:15:04 JSON API v2 initialized at /api/v2
HTTP server started on port 8080 (AutoTLS: false)
2025/04/25 15:15:04 Clip retention policy: usage
⇨ http server started on [::]:8080
2025/04/25 15:15:04 ⚠️ Audio device validation failed: configured audio device 'WordForum USB    Mono' not found

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ X] PC/Server (Late 2012 iMac Core i7)
- OS: Linux Mint 22.1 Cinammon
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: Intel© Core™ i7-3770 CPU @ 3.40GHz × 4
  - RAM: 32GB
  - Storage type and size: 2 TB SSD
- Additional hardware (if applicable):
  - USB Audio device: Word Forum USB Mono
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ X] No
- [ X] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
Have to run the standalone version, as install.sh claims my CPU is too old (it might be but probably isn't.)

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2025-04-25:
If you edit ~/.config/birdnet-go/config.yaml and put just "USB" as value for source, does it work then?

Like this

```yaml
realtime:
    interval: 15
    processingtime: false
    audio:
        source: USB
```

BirdNET-Go matches substrings on audio devices so this should match for "Word Forum USB Mono".

Can you also provide what arecord -l output looks like on your system?

```bash
arecord -l
```

tphakala, 2025-05-24:
Closing issue as there has been no responses

closed 2025-05-24 by tphakala