tphakala/birdnet-go #679: Error Occurred: Buffer write error for stream

J-shw, 2025-04-30:
## Issue Type
- [ ] Bug report
- [ ] Feature request
- [x] Question / Support

## Description
[A clear and concise description of the issue or feature request]
I'm not sure if this is a bug or incorrect setup on my end. But I keep getting issues with my RTSP stream running out of page file and causing this "Buffer write error for stream" issue

It seems to cause the sound recordings to be discarded and missing lots of bird detections because of this.

## Steps to Reproduce (for bugs)
1. [First Step]
2. [Second Step]
3. [and so on...]

## Expected behavior
[What you expected to happen]
Bird detections to be saved and analysed
## Actual behavior
[What actually happened]
Error message gets thrown in command line + debug report created in config folder - Audio gets discarded and bird detections ignored
## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```
Not sure how to run this command while its in docker. Web ui shows `nightly-20250315 `

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [x] PC/Server (please specify)
- OS: Ubuntu
- OS Version: Ubuntu 24.04.2 LTS 
- Hardware specs:
  - CPU: 11th Gen Intel(R) Core(TM) i5-11320H @ 3.20GHz (4c/8t)
  - RAM: 16GB
  - Storage type and size: 128GB SSD

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [x] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)
Running nightly, causes the container to fail to start 

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]
```
======== DEBUG INFO START ========
Error Occurred: Buffer write error for stream rtsp://192.168.86.105:554/h264_stream
CPU Utilization: 53.77%
RAM Usage: 71.14%
Page File Usage: 100.00%

Process List (ps axuw):
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           1  4.8  1.3 3259536 222192 ?      Ssl  07:19   1:26 /usr/bin/birdnet-go realtime
root          19  1.8  0.0 335648 14580 ?        SLl  07:19   0:33 ffmpeg -rtsp_transport tcp -i rtsp://192.168.86.105:554/h264_stream -loglevel error -vn -f s16le -ar 48000 -ac 1 -hide_banner pipe:1
root          58  0.0  0.0   8104  4096 ?        R    07:49   0:00 ps axuww
Go Runtime: Alloc = 16 MiB, TotalAlloc = 3860 MiB, Sys = 139 MiB, NumGC = 355
======== DEBUG INFO END ========
```

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2025-04-30:
Could you please run support script (instructions here https://github.com/tphakala/birdnet-go/wiki/BirdNET%E2%80%90Go-Guide#support-script) and attach file that is created to this ticket so I could take a closer look what is happening?

J-shw, 2025-04-30:
I have left the support script running for a few minutes and it seems to be stuck here: (I have stopped and started it a few times too)

```
 ____  _         _ _   _ _____ _____    ____      
| __ )(_)_ __ __| | \ | | ____|_   _|  / ___| ___ 
|  _ \| | '__/ _` |  \| |  _|   | |   | |  _ / _ \
| |_) | | | | (_| | |\  | |___  | |   | |_| | (_) |
|____/|_|_|  \__,_|_| \_|_____| |_|    \____|\___/ 

Support Data Collection Tool

📁 Created temporary directory for diagnostic data: /tmp/tmp.tSu0pt2TwO
🚀 Starting data collection process...
🖥️  Collecting system information...
🐳 Collecting Docker information...
📄 Collecting BirdNET-Go configuration...
```

I also restarted my birdnet-go container and get these errors:

```
birdnet-go  | 2025/04/30 14:30:04 ❌ Unexpected error writing to analysis buffer for stream rtsp://192.168.86.105:554/h264_stream: too many data to write
birdnet-go  | 2025/04/30 14:30:06 Abnormal event detected. Debug information written to: /root/.config/birdnet-go/debug_2025-04-30_14-30-06.txt
birdnet-go  | 2025/04/30 14:31:07 ⚠️ No data received from RTSP source rtsp://192.168.86.105:554/h264_stream for 60 seconds, triggering restart
```

With the following log file:

```
======== DEBUG INFO START ========
Error Occurred: Buffer write error for stream rtsp://192.168.86.105:554/h264_stream
CPU Utilization: 91.70%
RAM Usage: 77.67%
Page File Usage: 100.00%

Process List (ps axuw):
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           1 34.2  1.8 2464120 294836 ?      Ssl  14:29   0:18 /usr/bin/birdnet-go realtime
root          19  0.7  0.3 335676 50980 ?        SLl  14:29   0:00 ffmpeg -rtsp_transport tcp -i rtsp://192.168.86.105:554/h264_stream -loglevel error -vn -f s16le -ar 48000 -ac 1 -hide_banner pipe:1
root          33 66.6  0.0   8104  4096 ?        R    14:30   0:00 ps axuww
Go Runtime: Alloc = 12 MiB, TotalAlloc = 104 MiB, Sys = 38 MiB, NumGC = 13
======== DEBUG INFO END ========
```

I am going to do a sytem reboot and see if that helps sort this out..

J-shw, 2025-05-01:
Just a small update, the system has been running for around 15 hours now and no issues yet. I will close this issue for now. Thanks

closed 2025-05-01 by J-shw