tphakala/birdnet-go #686: install error on Nightly Build 20250427

itnetwork, 2025-05-01:
## Issue Type
- [x ] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Fresh install on proxmox updated Debian 12 lxc container.  Also tried Ubuntu (Same result)

## Steps to Reproduce (for bugs)
1. Use install.sh with the nightly tag. This also borked a legitimate install when updating

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ x] PC/Server (proxmox Debian and Ubuntu lxc container)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
🐳 Waiting for container to start...
⏳ Waiting for container to start (attempt 1/30)...
❌ Service stopped unexpectedly
Checking service logs:
May 01 17:17:17 birdnet-go systemd[1]: Starting birdnet-go.service - BirdNET-Go...
May 01 17:17:17 birdnet-go docker[3086]: Error: No such container: birdnet-go
May 01 17:17:17 birdnet-go systemd[1]: Started birdnet-go.service - BirdNET-Go.
May 01 17:17:17 birdnet-go docker[3095]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
May 01 17:17:17 birdnet-go systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
May 01 17:17:17 birdnet-go sh[3145]: umount: /root/birdnet-go-app/config/hls: block devices are not permitted on filesystem.
May 01 17:17:17 birdnet-go docker[3146]: Error: No such container: birdnet-go
May 01 17:17:17 birdnet-go systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
May 01 17:17:17 birdnet-go systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 1.
May 01 17:17:17 birdnet-go systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
May 01 17:17:17 birdnet-go systemd[1]: Starting birdnet-go.service - BirdNET-Go...
May 01 17:17:17 birdnet-go docker[3152]: Error: No such container: birdnet-go
May 01 17:17:17 birdnet-go systemd[1]: Started birdnet-go.service - BirdNET-Go.
May 01 17:17:18 birdnet-go docker[3162]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.
May 01 17:17:18 birdnet-go systemd[1]: birdnet-go.service: Main process exited, code=exited, status=127/n/a
May 01 17:17:18 birdnet-go sh[3197]: umount: /root/birdnet-go-app/config/hls: block devices are not permitted on filesystem.
May 01 17:17:18 birdnet-go docker[3198]: Error: No such container: birdnet-go
May 01 17:17:18 birdnet-go systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
May 01 17:17:18 birdnet-go systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 2.
May 01 17:17:18 birdnet-go systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
May 01 17:17:18 birdnet-go systemd[1]: Starting birdnet-go.service - BirdNET-Go...

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

collinjackson93, 2025-05-04:
I'm encountering the same issue on a Debian 12 lxc in Proxmox 8.3.4. I can't tell if the primary issue is the docker container not existing (`docker[3086]: Error: No such container: birdnet-go`) or attempting to access the sound device (`docker[3095]: docker: Error response from daemon: error gathering device information while adding custom device "/dev/snd": no such file or directory.`)

In case it matters, when setting up the config via the install script, I selected an rtsp stream as the input rather than a local device.

tphakala, 2025-05-05:
Reason is there in the error message, docker refuses to start as it cannot map /dev/snd device to birdnet-go container. 

You have two options

1. Remove /dev/snd mapping from birdnet-go systemd start script `/etc/systemd/system/birdnet-go.service`
2. Add sound device to lxc container, doesn't have to be real audio device, just something to make /dev/snd appear on container host

collinjackson93, 2025-05-05:
I confirmed that removing the mapping from the systemd file resolved the issue. I appreciate the speedy response!

itnetwork, 2025-05-06:
Me too. Thanks Tomi. Love the new analytics.

closed 2025-05-06 by itnetwork