tphakala/birdnet-go #696: Results queue is full! - how to clean?

simplemice, 2025-05-03:
fresh install on lxc container in proxmox

`birdnet-go  | 2025/05/03 07:49:01 🧹 Running clip cleanup task
birdnet-go  | 2025/05/03 07:49:01 🧹 Usage-based cleanup completed successfully, clips removed: 0, current disk utilization: 62%
birdnet-go  | 2025/05/03 07:49:12 🔊 Active audio broadcast callbacks: []
birdnet-go  | 2025/05/03 07:49:12 ⚠️ No broadcast callback registered for source: rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0, data length: 15360 bytes
birdnet-go  | 2025/05/03 07:49:47 ✅ TCP Connection: Running TCP Connection test...
birdnet-go  | 2025/05/03 07:49:47 ✅ TCP Connection: Successfully completed TCP Connection
birdnet-go  | 2025/05/03 07:49:47 ✅ MQTT Connection: Running MQTT Connection test...
birdnet-go  | 2025/05/03 07:49:47 ✅ MQTT Connection: Successfully completed MQTT Connection
birdnet-go  | 2025/05/03 07:49:47 ✅ Message Publishing: Running Message Publishing test...
birdnet-go  | 2025/05/03 07:49:52 ❌ Message Publishing: Message Publishing operation timed out: operation timeout
birdnet-go  | 2025/05/03 07:50:43 Human detected with confidence 0.106/0.050 from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0
birdnet-go  | 2025/05/03 07:51:21 Human detected with confidence 0.050/0.050 from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0
birdnet-go  | 2025/05/03 07:51:31 Human detected with confidence 0.052/0.050 from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0
birdnet-go  | 2025/05/03 07:52:00 Human detected with confidence 0.058/0.050 from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0
birdnet-go  | 2025/05/03 07:52:49 Discarding detection of asian koel from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0 due to false positive, matched 1/2 times
birdnet-go  | 2025/05/03 07:53:03 Approving detection of asian koel from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0, matched 2 times
birdnet-go  | 2025/05/03 07:53:19 Approving detection of asian koel from source rtsp://rtsp:rtsp@192.168.3.173:554/av_stream/ch0, matched 2 times
birdnet-go  | 2025/05/03 07:53:29 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:29 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:32 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:32 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:34 ⚠️ Job 4050220b: Publish detection to MQTT failed, will retry in 20.346954468s (attempt 1/4): job execution timed out after 30 seconds: context deadline exceeded
birdnet-go  | 2025/05/03 07:53:35 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:35 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:38 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:38 ❌ Results queue is full!
birdnet-go  | 2025/05/03 07:53:39 ❌ Results queue is full!
`

tphakala, 2025-05-03:
"Publish detection to MQTT failed" - you need to fix failing MQTT publish service, either your MQTT broker isn't accepting connections from BirdNET-Go or you have incorrect broker address etc. "Results queue is full" error is result of MQTT publish action failing.

tphakala, 2025-05-03:
Also, when filing issue ticket, please fill in issue template, it is there for a purpose as I do need a certain basic information to be able to help effectively.

closed 2025-05-03 by simplemice