tphakala/birdnet-go #737: RTSP doesn't reconnect

VerbalP, 2025-05-28:
Please fill in this issue template, any issue ticket filed without basic information required in this template will be closed without response. I cannot help if I don't have basic details like application version etc.

## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Please fill in this issue template, any issue ticket filed without basic information required in this template will be closed without response. I cannot help if I don't have basic details like application version etc.

## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
This is a follow-up to https://github.com/tphakala/birdnet-go/issues/558 (I'm not allowed to reopen it)

I have the same issue / the bug is still present on main branch 

## Steps to Reproduce (for bugs)
Cut the RTSP stream for a while then have it back online

## Expected behavior
Reconnects
## Actual behavior
Doesn't reconnect

## Environment
### Version Information (build date)
```
BirdNET-Go nightly-20250512 (built: 2025-05-23T00:32:42Z)
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server (please specify)
- OS: Docker on Debian

## Possible Solution
@aav7fl mentions in https://github.com/tphakala/birdnet-go/issues/558#issuecomment-2766249717 that https://github.com/tphakala/birdnet-go/pull/566 seems to have resolved their connection issue. Would it be possible to merge (partially?) with main?

aav7fl, 2025-06-19:
@tphakala I took a look at the latest build (after you merged https://github.com/tphakala/birdnet-go/pull/761)

I tried it out from this image here: https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go/442141074?tag=ceb0c44108f1d8eddd0f065a9d8d09a3bd0c52e1

I started up BirdNET-Go. Once I confirmed it everything was running, I restarted one of my cameras and I received this log message in BirdNET-Go:

```bash
2025/06/19 08:57:52 ⚠️ No data received from RTSP source rtsp://REDACTED for 60 seconds, triggering restart
```

The RTSP warning was only logged once. I don't think my camera had finished restarting so it wasn't ready when BirdNET-Go attempted to connect to it again, which means the ffmpeg stream was never retried again. 

I waited ~10 minutes and it never attempted to reconnect again. The stream remained dead until I manually restarted BirdNET-Go.

aav7fl, 2025-06-20:
It looks like there were a few more commits in the last day with other improvements to the ffmpeg restart (like adding unlimited retry).

I tried pulling down this build: https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go/442996398?tag=ee657496f979c83659c74161754da0969d6ce1de

Which was built from this commit: https://github.com/tphakala/birdnet-go/commit/ee657496f979c83659c74161754da0969d6ce1de

But it appears that didn't work for me either. I only have the 1 log message for the ffmpeg process failing. This happens when my camera restarts.

```
2025/06/20 08:19:50 ⚠️ No data received from RTSP source REDACTED for 60 seconds, triggering restart
```

So something changed with how it is reconnecting. Maybe the request is hanging so it doesn't get a proper exit code, which prevents it from restarting the re-connection attempt?

aav7fl, 2025-06-28:
I tried the latest nightly https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250628 

I was hoping the ffmpeg fix included in a recent merge would have fixed it. 

I started up the nightly and confirmed all of my streams as working. Then I restarted my camera.

I saw one log message in attempting to recover the rtsp stream. Unforunately, my camera isn't yet finished rebooting when that message is sent, and the ffmpeg is never retried again. 

```
2025/06/28 14:13:26 ⚠️ No data received from RTSP source rtsp://REDACTED for 60 seconds, triggering restart
2025/06/28 14:14:31 🧹 Running HLS stream cleanup task
2025/06/28 14:14:31 ✅ HLS cleanup task completed: no active streams
2025/06/28 14:14:31 🧹 Running clip cleanup task
2025/06/28 14:14:33 🧹 Age-based cleanup completed successfully, clips removed: 5, current disk utilization: 50%
2025/06/28 14:14:35 🔊 Active audio broadcast callbacks: []
2025/06/28 14:14:35 ⚠️ No broadcast callback registered for source: rtsp://REDACTED, data length: 1440 bytes
2025/06/28 14:15:41 Retrieved 5 detections in 2.452267ms
2025/06/28 14:15:41 Total time for processing all notes: 22.183244ms
2025/06/28 14:17:26 Debug: ✅ CSRF token found in context for settings page: /settings/audio
Devices: [{1 HDA Intel PCH, ALC233 Analog :0,0}]
2025/06/28 14:17:27 Debug: ✅ CSRF token found in context for settings page: /settings/main
2025/06/28 14:17:30 Debug: ✅ CSRF token found in context for settings page: /settings/support
2025/06/28 14:18:10 Dog detected with confidence 0.139/0.100 from source rtsp://REDACTED
2025/06/28 14:19:31 🧹 Running clip cleanup task
2025/06/28 14:19:31 🧹 Running HLS stream cleanup task
2025/06/28 14:19:31 ✅ HLS cleanup task completed: no active streams
2025/06/28 14:19:33 🧹 Age-based cleanup completed successfully, clips removed: 9, current disk utilization: 50%
2025/06/28 14:19:35 🔊 Active audio broadcast callbacks: []
2025/06/28 14:19:35 ⚠️ No broadcast callback registered for source: rtsp://REDACTED, data length: 1440 bytes
2025/06/28 14:19:53 Dog detected with confidence 0.224/0.100 from source rtsp://REDACTED
```

I can see that BirdNET-Go knows the stream is silent based on this screenshot.

[image]

So there must be some other condition that's preventing it from retrying. Maybe since it knows the stream is silent, that would indicate a stream problem?

tphakala, 2025-06-28:
Sorry about this, I will work on this tomorrow.

aav7fl, 2025-07-10:
This is resolved for me based on conversations here: https://github.com/tphakala/birdnet-go/issues/811#issuecomment-3056616372

The cause was likely related to cameras holding their TCP connections upon restarting. By adding a socket timeout, BirdNET-Go can properly detect this and have ffmpeg attempt a reconnection. 

Resolved by: https://github.com/tphakala/birdnet-go/pull/917

MikkelPaulson, 2025-07-15:
Thank you! I have also been experiencing this and will check back in a few days to confirm if it's fixed on the latest nightly. (It's an intermittent issue and I haven't been able to reproduce it manually.) For what it's worth, I have Birdnet-pi running on the same machine + stream for Reasons, and its healthcheck also fails at exactly the same time. I'll be able to confirm next time its healthcheck fails if Birdnet-pi is still up.

MikkelPaulson, 2025-07-18:
Update: In fact, it's worse now. -Pi did in fact go down, but -Go crashed first this time. Something seems fishy with the "inactive for 0s" message, which it never seems to recover from.

```
birdnet-go  | 2025/07/16 14:51:14 🎧 Registered audio callback for source: rtsp://xxx:xxx@10.4.0.21/live0, total callbacks: 1
birdnet-go  | 2025/07/16 14:51:14 🎧 Registered audio callback for source: rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:51:14 ✅ Audio feed ready for source rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:51:15 ⏳ Waiting for playlist file (attempt 2): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:16 ⏳ Waiting for playlist file (attempt 3): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:17 ⏳ Waiting for playlist file (attempt 4): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:18 ⏳ Waiting for playlist file (attempt 5): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:18 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m13.069693615s ago)
birdnet-go  | 2025/07/16 14:51:18 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:51:19 ⏳ Waiting for playlist file (attempt 6): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:20 ⏳ Waiting for playlist file (attempt 7): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:21 ⏳ Waiting for playlist file (attempt 8): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:22 ⏳ Waiting for playlist file (attempt 9): /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playli
st.m3u8
birdnet-go  | 2025/07/16 14:51:23 ⚠️ Playlist check cancelled or timed out for source: rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:51:23 ⚠️ Playlist file not immediately available: /home/birdnet/.config/birdnet-go/hls/stream_350r1vrWpi1GRt08SFExd84D6DtDhvbQENEKgzDBPbs/playlist.m3u8
birdnet-go  | 2025/07/16 14:51:48 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m43.07045247s ago)
birdnet-go  | 2025/07/16 14:51:48 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:52:05 ✅ FFmpeg started for rtsp://10.4.0.21 (PID: 2634)
birdnet-go  | 2025/07/16 14:52:08 🔊 Active audio broadcast callbacks: [rtsp://xxx:xxx@10.4.0.21/live0]
birdnet-go  | 2025/07/16 14:52:08 ✅ First audio data successfully written to FIFO for source rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:52:08 🔄 Restart requested for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:52:08 🛑 FFmpeg process stopped for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:52:08 ✅ FFmpeg process ended normally for rtsp://10.4.0.21 after 2.851455161s
birdnet-go  | 2025/07/16 14:52:08 ⏳ Waiting 2m0s before restart attempt #479 for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:53:18 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m10.219153884s ago)
birdnet-go  | 2025/07/16 14:53:18 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:53:48 🧹 Running HLS stream cleanup task
birdnet-go  | 2025/07/16 14:53:48 ✅ HLS cleanup task completed: 0/1 streams cleaned up
birdnet-go  | 2025/07/16 14:53:48 📊 Oldest active stream: rtsp://10.4.0.21 (inactive for 2m34.492546774s)
birdnet-go  | 2025/07/16 14:53:48 🧹 Running clip cleanup task
birdnet-go  | 2025/07/16 14:53:48 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m40.219398355s ago)
birdnet-go  | 2025/07/16 14:53:48 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:53:48 🧹 Usage-based cleanup completed successfully, clips removed: 0, current disk utilization: 21%
birdnet-go  | 2025/07/16 14:54:08 ✅ FFmpeg started for rtsp://10.4.0.21 (PID: 2639)
birdnet-go  | 2025/07/16 14:54:10 🔄 Restart requested for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:54:10 🛑 FFmpeg process stopped for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:54:10 ✅ FFmpeg process ended normally for rtsp://10.4.0.21 after 2.330157659s
birdnet-go  | 2025/07/16 14:54:10 ⏳ Waiting 2m0s before restart attempt #480 for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:55:18 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m7.884096599s ago)
birdnet-go  | 2025/07/16 14:55:18 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:55:48 ⚠️ Unhealthy stream detected: rtsp://10.4.0.21 (last data: 1m37.884250608s ago)
birdnet-go  | 2025/07/16 14:55:48 🔄 Restarted FFmpeg stream for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:56:10 ✅ FFmpeg started for rtsp://10.4.0.21 (PID: 2644)
birdnet-go  | 2025/07/16 14:56:13 🔄 Restart requested for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:56:13 🛑 FFmpeg process stopped for rtsp://10.4.0.21
birdnet-go  | 2025/07/16 14:56:13 ✅ FFmpeg process ended normally for rtsp://10.4.0.21 after 2.404324282s
birdnet-go  | 2025/07/16 14:56:13 ⏳ Waiting 2m0s before restart attempt #481 for rtsp://10.4.0.21
[... cut ...]
birdnet-go  | 2025/07/16 14:56:22 ⚠️ Stream rtsp://xxx:xxx@10.4.0.21/live0 inactive for 0s (exceeds 5m0s timeout), marking for cleanup. Client count: 2
birdnet-go  | 2025/07/16 14:56:22 ⏳ Skipping cleanup of new HLS stream for source rtsp://2CreE5rbWt4gkGWH:u2Us5GqjwVUnWGoU@10.4.0.21/live0 (age: 0s)
birdnet-go  | 2025/07/16 14:56:27 ⚠️ Stream rtsp://xxx:xxx@10.4.0.21/live0 inactive for 0s (exceeds 5m0s timeout), marking for cleanup. Client count: 2
birdnet-go  | 2025/07/16 14:56:27 ⏳ Skipping cleanup of new HLS stream for source rtsp://2CreE5rbWt4gkGWH:u2Us5GqjwVUnWGoU@10.4.0.21/live0 (age: 0s)
[repeats every 5 seconds from this point on]
```

tphakala, 2025-07-22:
There was indeed remaining issues, can you please test latest release https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250722

mnhn-paul, 2025-08-10:
Was this fixed, as I suspect that I have the same issue. Only switched to Birdnetgo (on a vm) in the past days and my audio stream is unfortunately breaking once or twice a day...
submitted a report
2249-B5C0-FCEE

VerbalP, 2025-08-14:
Sadly, the issue appears to still be present on my end, using the docker image with nightly-20250728

VerbalP, 2025-08-14:
> [@VerbalP](https://github.com/VerbalP) please update to latest nightly release, also make sure to download new copy of install.sh as it will disable WIFI power saving on Raspberry PI hardware which may contribute to unstable RSTP connections.

Thanks @tphakala !
Updated now (docker image, no install), will report back in a few days ;-)

MikkelPaulson, 2025-09-02:
Okay, `nightly-20250813-1-ge680d30f` did indeed crash (and fail to come back) after a few days. In [logs.txt](https://github.com/user-attachments/files/22100849/logs.txt), the crash occurs at 03:09:50. I'll update to the latest -Go nightly and restart -Pi for good measure.

tphakala, 2025-09-04:
@MikkelPaulson what RTSP audio source do you have? Is it a Unifi camera by any chance?

MikkelPaulson, 2025-09-05:
No, it's a Eufy Outdoor Cam Pro. To be clear, the camera is wireless; the Pi is connected to the router by ethernet. So wifi might be *an* issue, but the Pi's wifi configuration is not *the* issue.

Ideally, the stream would never end up in an unrecoverable state that requires a manual restart, no matter how dodgy the network may be. So if anything, optimizing for network reliability makes it harder to track down the root cause, since no network will be 100% reliable.

(Still going strong since Tuesday, BTW.)