tphakala/birdnet-go #788: Add sound level reporting in 1/3rd octave

mfortini, 2025-06-26:
## Issue Type
- [ ] Bug report
- [x] Feature request
- [ ] Question / Support

## Description

While the system is capturing sound, it would be nice if it could be used also for reporting sound levels in 1/3rd octave intervals, perhaps every 10s or so (maybe also with 1s windows so that it would report min/max/mean in 10s).

The destination could be a JSON message in MQTT.

This way, a birdnet-go device could serve both as a bird detector and noise level sensor. Of course the sensor would not be calibrated, so that it wouldn't give out absolute values, but relative variations would be useful as well.

mfortini, 2025-06-27:
```
2025/06/27 20:07:43 🔊 Registered sound level processor for audio device: USB Audio Device, USB Audio
2025/06/27 20:07:43 ✅ Sound level monitoring enabled
2025/06/27 20:07:43 📊 Started sound level metrics publisher
2025/06/27 20:07:43 📡 Started sound level SSE publisher
2025/06/27 20:07:43 📡 Started sound level MQTT publisher
2025/06/27 20:07:44 Human detected with confidence 0.011/0.010 from source malgo
2025/06/27 20:07:52 Human detected with confidence 0.020/0.010 from source malgo
2025/06/27 20:07:53 Human detected with confidence 0.046/0.010 from source malgo
2025/06/27 20:07:54 Human detected with confidence 0.079/0.010 from source malgo
2025/06/27 20:07:55 Human detected with confidence 0.338/0.010 from source malgo
2025/06/27 20:07:56 Human detected with confidence 0.258/0.010 from source malgo
2025/06/27 20:07:57 Human detected with confidence 0.176/0.010 from source malgo
2025/06/27 20:07:58 Human detected with confidence 0.243/0.010 from source malgo
2025/06/27 20:07:59 Human detected with confidence 0.047/0.010 from source malgo
2025/06/27 20:08:00 Human detected with confidence 0.015/0.010 from source malgo
2025/06/27 20:08:01 Human detected with confidence 0.143/0.010 from source malgo
2025/06/27 20:08:02 Human detected with confidence 0.034/0.010 from source malgo
2025/06/27 20:08:03 Human detected with confidence 0.395/0.010 from source malgo
2025/06/27 20:08:04 Human detected with confidence 0.019/0.010 from source malgo
2025/06/27 20:08:05 Human detected with confidence 0.092/0.010 from source malgo
2025/06/27 20:08:06 Human detected with confidence 0.048/0.010 from source malgo
2025/06/27 20:08:07 Human detected with confidence 0.069/0.010 from source malgo
2025/06/27 20:08:09 Human detected with confidence 0.040/0.010 from source malgo
2025/06/27 20:08:10 Human detected with confidence 0.011/0.010 from source malgo
2025/06/27 20:08:11 Human detected with confidence 0.020/0.010 from source malgo
2025/06/27 20:08:13 Human detected with confidence 0.014/0.010 from source malgo
2025/06/27 20:08:13 ❌ Error publishing sound level data to MQTT: json: unsupported value: +Inf
```

tphakala, 2025-06-28:
I applied some more fixes, now it should be working at least on some level. https://github.com/tphakala/birdnet-go/pull/804

Please test and provide feedback

mfortini, 2025-06-29:
Now the calculation is done and send via MQTT, great

tphakala, 2025-06-30:
The sound level measurement system reads the raw audio straight from the source - no EQ, no filtering, just pure unprocessed audio.

closed 2025-07-08 by mfortini