tphakala/birdnet-go #881: Add push notification support for errors now being reported

oester, 2025-07-06:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type
- [ ] Bug report
- [ X] Feature request
- [ ] Question / Support

## Description
When a problem is detected and reported via the “bell” icon, the issue can go unnoticed for a long period. When it’s an error associated with something like mqtt or rtsp, it would be good to be notified. The “apprise” notification package is straightforward to include and covers a wide range of notification types. Implementation would be ideal if you could filter by event types or severity. The use of this package is a suggestion and could be implemented in other ways.

tphakala, 2025-07-06:
I agree that it would be useful. I filed somewhat detailed implementation plan https://github.com/tphakala/birdnet-go/issues/882, I would be glad if someone else would start working on implementing this as I need to get audio core rewrite complete in reasonable time.

oester, 2025-07-06:
Thanks, that looks excellent.  And thanks to you for all the hard work in making birdnet-go such a fantastic piece of software!