tphakala/birdnet-go #887: Feature: Onboarding Wizard for New Users

tphakala, 2025-07-06:
## Summary

Implement a comprehensive onboarding wizard that guides new BirdNET-Go users through initial setup, educates them about AI limitations, and provides an interactive tour of the web interface. This will significantly improve the first-time user experience and reduce configuration errors.

**📝 Note: This feature proposal is open for comments and feedback from the community. Please share your thoughts on the design, suggest improvements, or indicate which aspects would be most valuable to you.**

## Background

New users currently need to manually configure BirdNET-Go through YAML files or the web interface without guided assistance. This can be overwhelming and may lead to suboptimal configurations or missed features. An onboarding wizard would:

- Guide users through essential configuration steps
- Educate about AI model limitations and appropriate use cases
- Showcase key features through an interactive tour
- Provide option to import existing BirdNET-Pi data
- Ensure users understand privacy implications and data sharing options

## Proposed Solution

### 1. Welcome Screen & AI Disclaimer

On first launch, display a modal dialog with:

```
Welcome to BirdNET-Go\!

Before we begin, please understand:
• BirdNET AI model may make identification mistakes
• Detections should not be considered scientifically accurate without verification
• Best suited for hobby observation and citizen science
• For research purposes, always verify identifications manually

[✓] I understand these limitations

[Continue] [Import from BirdNET-Pi]
```

### 2. Setup Wizard (5 Steps)

#### Step 1: Location Configuration
- Interactive map widget for setting latitude/longitude
- Auto-detect location option (with permission)
- Display: "Your location helps filter species to those likely in your area"
- Show preview of common species for the selected location

#### Step 2: Audio Source
- Radio buttons for audio source selection:
  - 🎤 Microphone/Soundcard (default)
  - 📹 RTSP Stream
  - 🔊 Multiple Sources
- Audio level meter showing real-time input
- "Test Audio" button with visual feedback
- Troubleshooting link if no audio detected

#### Step 3: Language & Display
- Dropdown with 40+ language options for species names
- Time format toggle (12/24 hour)
- Timezone selection (auto-detect with override option)
- Preview of species name in selected language

#### Step 4: Detection Settings
- Preset selection with descriptions:
  ```
  ⚖️ Balanced (Recommended)
  Good balance of accuracy and sensitivity
  • Confidence threshold: 0.8
  • Sensitivity: 1.0
  
  🎯 High Accuracy
  Fewer false positives, may miss some birds
  • Confidence threshold: 0.9
  • Deep Detection enabled
  
  📡 High Sensitivity  
  Detects more birds, more false positives
  • Confidence threshold: 0.6
  • Sensitivity: 1.25
  ```
- "Show Advanced Settings" expandable section

#### Step 5: Privacy & Integration
- Privacy options:
  - [ ] Enable privacy filter (removes detections when human voices detected)
  - [ ] Enable dog bark filter (reduces false owl/crow detections)
- Data sharing:
  - [ ] Share detections with BirdWeather community
  - [ ] Enable anonymous telemetry to help improve BirdNET-Go
- Audio retention:
  - Export audio clips: [Yes/No]
  - Retention policy: [7 days / 30 days / Forever]

### 3. Interactive Web UI Tour

Using a JavaScript tour library (e.g., Shepherd.js), create guided tour with these stops:

1. **Dashboard Overview**
   ```
   Spotlight: Main dashboard area
   "This is your command center. Recent detections appear here
   in real-time. The summary shows your most detected species."
   ```

2. **Detection Details**
   ```
   Spotlight: Detection row
   "Click any detection to see the spectrogram, listen to audio,
   and verify or correct the identification."
   ```

3. **Live Audio Stream**
   ```
   Spotlight: Audio level indicator
   "Click here to listen to your live audio feed and check
   microphone placement."
   ```

4. **Settings Access**
   ```
   Spotlight: Settings menu
   "Fine-tune detection parameters, manage species lists,
   and configure integrations here."
   ```

5. **Species Management**
   ```
   Spotlight: Species settings
   "Add species to always include or exclude, and set
   custom thresholds for specific birds."
   ```

### 4. BirdNET-Pi Import

If user selects import option:

```
Import from BirdNET-Pi

1. Upload your BirdNET-Pi database:
   [Choose File] (Accepts: birds.db, *.sqlite)

2. Import options:
   [✓] Historical detections
   [✓] Species include/exclude lists
   [✓] Custom species settings
   [ ] Audio recordings (if available)

3. Date range:
   From: [Date picker] To: [Date picker]

[Import] [Skip]

Progress: [████████░░] 80% - Importing detections...
```

### 5. Completion Screen

```
✅ Setup Complete\!

BirdNET-Go is now monitoring for bird sounds.

Quick tips:
• First detections may take 15-30 seconds to appear
• Check Settings → Range Filter to see species expected in your area
• Enable notifications for rare bird alerts
• Join our community forum for tips and support

[View Dashboard] [Documentation] [Run Tour Again]
```

## Implementation Tasks

- [ ] Create wizard component structure in web UI
- [ ] Implement state management for wizard progress
- [ ] Add API endpoint: `GET /api/v2/wizard/status`
- [ ] Add API endpoint: `POST /api/v2/wizard/complete`
- [ ] Create location picker component with map
- [ ] Implement audio source selector with level meter
- [ ] Build preset configuration templates
- [ ] Integrate JavaScript tour library
- [ ] Create tour content and styling
- [ ] Implement BirdNET-Pi import functionality
- [ ] Add database migrations for import
- [ ] Create user preference storage for wizard completion
- [ ] Add "Reset Wizard" option in settings
- [ ] Implement skip/back navigation
- [ ] Add help tooltips and contextual information
- [ ] Create responsive mobile layout
- [ ] Write user documentation
- [ ] Add integration tests

## UI/UX Considerations

- Use consistent styling with existing BirdNET-Go interface
- Ensure accessibility (ARIA labels, keyboard navigation)
- Provide clear progress indication (step 1 of 5)
- Allow users to go back and modify previous choices
- Save progress in case of interruption
- Offer "Quick Setup" vs "Custom Setup" paths
- Include contextual help and tooltips
- Support for mobile and tablet layouts

## Technical Requirements

- Frontend: Leverage existing Alpine.js/HTMX stack
- Use Tailwind CSS for consistent styling  
- Store wizard completion state in user preferences
- Graceful fallback if JavaScript disabled
- Compatible with all authentication methods
- Internationalization support for wizard text

## Success Metrics

- Reduction in support requests about initial configuration
- Increased percentage of users enabling integrations
- Faster time from installation to first detection
- Higher user retention in first week
- Positive feedback on user experience

## Future Enhancements

- Video tutorials embedded in wizard steps
- Automatic hardware optimization based on system detection
- Community-contributed location-specific tips
- Integration with hardware setup (USB audio devices)
- Suggested microphone placement based on environment

## References

- Current installation documentation: `/doc/wiki/installation.md`
- Configuration guide: `/doc/wiki/guide.md`
- Similar implementations: 
  - Home Assistant onboarding
  - Nextcloud first-run wizard
  - Gitlab initial setup

## Request for Feedback

This is a proposal for improving the new user experience in BirdNET-Go. We'd love to hear from the community:

- Which features would be most valuable to you?
- Are there additional configuration steps that should be included?
- What challenges did you face when first setting up BirdNET-Go?
- Any suggestions for the UI/UX design?

Please share your thoughts in the comments below\!

Sridhar3145, 2025-07-27:
Hi @tphakala,

I'd love to contribute to the onboarding wizard implementation (#887) as a frontend developer. Could you assign me to the frontend-specific portions of this issue?

I can focus on:

Wizard UI components (Alpine.js/HTMX)

Interactive tour integration (Shepherd.js)

i18n implementation (#999) for the wizard

Responsive/mobile layouts

I may need backend support for:

API endpoints (/wizard/status, /wizard/complete)

BirdNET-Pi import functionality

Database migrations

Let me know if you'd like me to break down the frontend tasks into separate sub-issues for better tracking.

Thanks for considering!

tphakala, 2025-07-28:
Hi @Sridhar3145 thanks for reaching out!

I recenlty rewrote user interface in Svelte 5 and typescript, it is getting very close being feature complete with old HTMX and Alpine.js based one, at this point all new features and development effort should go to Svelte based UI.

If you have latest nightly release of BirdNET-Go deployd Svelte based ui is available at birdnet-go:8080/ui/ path, code is in frontend/ directory within repository. With this new UI there also will be completely separate desktop and mobile interfaces, I have not started work on mobile interface yet.

Please familiarise yourself with new UI and lets discuss more.

tphakala, 2026-01-15:
Closing - onboarding wizard deferred