tphakala/birdnet-go #966: Failed to perform Soundscape Upload failed to upload soundscape

mcox24680, 2025-07-17:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type
- [ X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
error enabling birdWeather

## Steps to Reproduce (for bugs)
1. enable birdWeather
2. type in token
3. see failure

## Expected behavior
login successful

## Actual behavior

Soundscape Upload
Failed to perform Soundscape Upload failed to upload soundscape: soundscape upload failed with status 415: {"success":true,"soundscape":{"id":null,"stationId":15588,"timestamp":"2025-07-17T05:19:50.405-07:00","url":null,"filesize":null,"extension":null,"duration":null,"mode":"birdnetpi"},"message":"WAV support has been discontinued and will no longer be stored. Please switch to FLAC as the soundscape audio format."} Failed to upload the test soundscape. This could be due to network issues, authentication problems, or the BirdWeather service might be experiencing issues. Try again later or check your BirdWeather station settings.

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ X] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
# Alternative: docker pull tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2025-07-17:
Which version of BirdNET-Go are you using? Looking at error message you may be using some old and already obsolete version. Please use latest nightly releases.

mcox24680, 2025-07-17:
i ran these 2 commands:

curl -fsSL https://github.com/tphakala/birdnet-go/raw/main/install.sh -o install.sh
 bash ./install.sh

can i provide any other info help?

jmiller0, 2025-07-17:
I just installed today for the first time (Awesome project) running docker build nightly-20250710-102-g62e4b8c6.  I have the exact same error:

`Soundscape Upload
Failed to perform Soundscape Upload failed to upload soundscape: soundscape upload failed with status 415: {"success":true,"soundscape":{"id":null,"stationId":15591,"timestamp":"2025-07-17T07:09:09.323-07:00","url":null,"filesize":null,"extension":null,"duration":null,"mode":"birdnetpi"},"message":"WAV support has been discontinued and will no longer be stored. Please switch to FLAC as the soundscape audio format."} Failed to upload the test soundscape. This could be due to network issues, authentication problems, or the BirdWeather service might be experiencing issues. Try again later or check your BirdWeather station settings.`

I changed the record settings to flac after setting up the birdweather integration but still getting this error, so I'm guessing its testing with a old wav file?  Can I just delete the old wav files or does it need to be done in the db?  Or does the test use something else?

[image]

Thanks

jmiller0, 2025-07-17:
I generated the support report and hit 'Generate and Upload' it looks like it saved locally, does it upload automatically?  Nothing in the log gets printed that I can see.. (I referenced this github and my System ID)

mcox24680, 2025-07-17:
would you like my logs as well?

whiney1, 2025-07-18:
chiming in to say I have the same issue - running addon via HomeAssistant Addon, version nightly-20250710

same error message when testing BirdWeather integration, audio capture set to FLAC etc.

tphakala, 2025-07-18:
@mcox24680 @whiney1 please do send support bundles, I need a lot more context why this is happening.

Make sure to enable error reporting on support page first, also it might be a good idea to restart BirdNET-Go after you have enabled error reporting and reload page fully before generating support bundle to make sure it works.

tphakala, 2025-07-18:
Ah hey one more question, is this happening only when you do "Test BirdWeather Connection" in Settings UI, or do you see these errors in logs/birdweather.log also?

Because I just realized that this issue indeed happens with test, but actual submissions do happen with FLAC format. I need to update test to submit FLAC type audio file as well.

whiney1, 2025-07-18:
Done.
Just checked and my uploads to BirdWeather are actually working, so does seem to just be a test issue.

Error does appear in my local logs (homeassistant addon).

tphakala, 2025-07-18:
> Upload seems to have failed, even though it was successful in BN-G, similar to the guy above:

I disabled Automated Support Engineer workflow for now, should stop these errors.