tphakala/birdnet-go #977: Trying to install on new windows PC but can't get to work

pack2010, 2025-07-19:
Hello!

I had birdnet installed on my old PC and it ran with no issues but I can't seem to get the new one to work properly. When running it in CMD I get this sequence:

2025/07/19 12:14:27 SoX not found in system PATH. Audio source processing requiring SoX might be disabled.
🐦 BirdNET-Go 6b9790f (built: 2025-07-18T18:30:28Z), using config file: D:\birdnet\config.yaml
2025/07/19 12:14:27 INFO starting core services initialization service=system-init
2025/07/19 12:14:27 INFO starting telemetry initialization sequence service=init-coordinator
2025/07/19 12:14:27 INFO error integration initialized successfully service=init-manager
2025/07/19 12:14:27 INFO telemetry initialization sequence completed (event bus integration deferred) service=init-coordinator
2025/07/19 12:14:27 INFO notification service initialized successfully service=system-init debug=false
2025/07/19 12:14:27 INFO core services initialization completed service=system-init
Warning: core systems initialization incomplete: telemetry not initialized
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 12 threads of available 12 CPUs
2025/07/19 12:14:27 🧹 Cleaned up leftover HLS streaming files
System details: windows Microsoft Windows 10 Home 10.0.19045.6093 Build 19045.6093 on unknown hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 1.5, sensitivity: 1, interval: 15
2025/07/19 12:14:27 Using global image provider registry
2025/07/19 12:14:27 Registered WikiMedia image provider
2025/07/19 12:14:27 Attempting to register AviCommons provider...
2025/07/19 12:14:27 Successfully registered AviCommons image provider
2025/07/19 12:14:27 Using WikiMedia as the default image provider (auto mode)
2025/07/19 12:14:27 Starting background image cache warm-up...
2025/07/19 12:14:27 Pre-fetched 0 cached image records for provider 'wikimedia'
2025/07/19 12:14:27 Pre-fetched 0 cached image records for provider 'avicommons'
2025/07/19 12:14:27 BirdImageCache initialized (all species images already present in DB cache)
2025/07/19 12:14:27 Job queue started with max capacity of 1000 jobs
2025/07/19 12:14:27 Error: Backup logger is nil. Logging may not be initialized.
2025/07/19 12:14:27 INFO Initializing backup system...
2025/07/19 12:14:27 INFO No existing backup state file found, creating new one service=backup_statemanager path=D:\birdnet\backup-state.json
2025/07/19 12:14:27 INFO Backup system is disabled.
2025/07/19 12:14:27 INFO Backup system initialized.
2025/07/19 12:14:27 INFO starting async services initialization service=system-init
2025/07/19 12:14:27 INFO event bus initialized successfully service=system-init buffer_size=10000 workers=4
2025/07/19 12:14:27 INFO initializing notification event bus integration service=notification-integration
2025/07/19 12:14:27 INFO notification worker registered with event bus service=notification-integration consumer=notification-worker supports_batching=false batching_enabled=false batch_size=10 batch_timeout=100ms circuit_breaker_threshold=5 recovery_timeout=30s debug=false
2025/07/19 12:14:27 INFO resource worker registered with event bus service=notification-integration consumer=notification-resource-worker alert_throttle=5m0s debug=false
2025/07/19 12:14:27 INFO notification worker initialized successfully service=system-init
2025/07/19 12:14:27 ERROR telemetry event bus initialization failed service=system-init error="logging not initialized"
2025/07/19 12:14:27 INFO async services initialization completed service=system-init
2025/07/19 12:14:27 INFO initializeSystemMonitor called monitoring_enabled=true check_interval=60
2025/07/19 12:14:27 INFO Creating system monitor instance
Error opening /proc/self/cgroup: open /proc/self/cgroup: The system cannot find the path specified.
2025/07/19 12:14:27 INFO Starting system monitor
2025/07/19 12:14:27 INFO System resource monitoring initialized component=monitor interval=60
Logging disabled
2025/07/19 12:14:27 Configured custom IP extractor prioritizing CF-Connecting-IP
2025/07/19 12:14:27 Resolved relative media export path "clips/" to absolute path "D:\birdnet\clips"
2025/07/19 12:14:27 API structured logging initialized to logs/web.log
2025/07/19 12:14:27 Initialized API authentication service and middleware function
2025/07/19 12:14:27 Debug mode not enabled, skipping debug routes
HTTP server started on port 8080
2025/07/19 12:14:27 🔇 Sound level monitoring disabled
2025/07/19 12:14:27 🔍 RTSP streams will be monitored by FFmpeg manager
2025/07/19 12:14:27 ✅ Started FFmpeg stream for rtsp://***.***.*.***:***
2025/07/19 12:14:27 Clip retention policy: usage
⇨ http server started on [::]:8080
2025/07/19 12:14:27 ✅ FFmpeg started for rtsp://***.***.*.***:*** (PID: 18732)
2025/07/19 12:14:27 ⚠️ Audio device validation failed: no hardware audio capture devices found
2025/07/19 12:14:30 🔊 Active audio broadcast callbacks: []
2025/07/19 12:14:30 ⚠️ No broadcast callback registered for source: rtsp://*****:**********@******.*.***:***/h264Preview_01_main, data length: 6048 bytes
2025/07/19 12:14:32 ✅ Stream rtsp://192.168.7.191:554 is healthy and receiving data (130.4 KB/s)
2025/07/19 12:15:27 🧹 Running HLS stream cleanup task
2025/07/19 12:15:27 ✅ HLS cleanup task completed: no active streams

tphakala, 2025-07-19:
There is lots of log but please describe your issue, what is not working exactly?

I see that application starts properly, http server is available at port 8080. You don't seem to have sound card though, "Audio device validation failed: no hardware audio capture devices found", you have one RTSP stream configured which seems to be working fine "rtsp://192.168.7.191:554". I don't see much wrong here, other than you don't have SoX installed which causes spectrograms not to appear.

tphakala, 2026-01-15:
Closing as stale - This support issue is 6 months old and the reporter hasn't provided additional details. If you're still experiencing Windows installation issues on the latest release, please open a new issue with:

1. Current BirdNET-Go version
2. Complete error logs
3. Configuration file (with sensitive data redacted)

closed 2026-01-15 by tphakala