tphakala/birdnet-go #982: Incorrect date/time displayed on search results

Lenbok, 2025-07-22:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

When I use the search page, the displayed times are off by 12 hours (e.g. they display as pm, when the result was am).

## Steps to Reproduce (for bugs)
1. Go the search page
2. Search for a species
3. Look at the date/time of the resulting hits

## Expected behavior

The correct date/time should be printed

## Actual behavior

The times all out by 12hr. 

My timezone is GMT+12,  and the displayed time is equivalent to GMT+24, perhaps the timezone offset is being double applied?

## Screenshots

Today's dashboard results for a species (these are correct):

[image]

Search results for the same species (these are incorrect):

[image]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
 nightly-20250710-102-g62e4b8c6 
```

### System Information
- Device: 
  - [x] Raspberry Pi 4 Model B
- OS:  Debian

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

Lenbok, 2025-07-29:
FYI, this is still an issue with the new UI, as of nightly-20250728

tphakala, 2025-07-30:
This is now fixed in https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250730

commit 99dd6547 (2025-07-30): fix: resolve timezone handling bugs in search and dashboard APIs (#1038)

* fix: resolve timezone handling bugs in search and dashboard APIs

- Fix timezone parsing in search results by using time.ParseInLocation() with time.Local
- Fix timezone parsing in analytics queries for dashboard summaries
- Fix timezone parsing in detection API responses
- Fix timezone parsing in template functions for date formatting
- Add comprehensive test suite for timezone edge cases

Database stores date/time as local time strings without timezone info.
Using time.Parse() incorrectly assumes UTC, causing 12+ hou [... cut]