tphakala/birdnet-go #1023: New UI dashboard won't let me see today

Lenbok, 2025-07-27:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

I'm not sure if  you want bug reports on the new UI yet - Feel free to tell me to wait until some kind of release if it's still too early days.

I thought I would try out the new UI. and noted first up that the dashboard is displaying the wrong day, and the ability to page to today is greyed out.

[image]

My timezone is UTC+12, so it may be that the UI is determining "today"  from GMT? The old UI doesn't have this issue.

## Steps to Reproduce (for bugs)
1. [First Step]
2. [Second Step]
3. [and so on...]

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
nightly-20250725-3-ge6edbb3d
```

### System Information
- Device: 
  - [x] Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)

tphakala, 2025-07-28:
Yes I do appreciate bug reports for new UI, this is one issue I had not spotted myself yet, so thank you for reporting!

Lenbok, 2025-07-28:
Nice! Will the timezone fixes have also addressed https://github.com/tphakala/birdnet-go/issues/982 ?
It would be great if there is a new nightly build to try!

tphakala, 2025-07-30:
This is now fixed in https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250730

commit 04de0ac1 (2025-07-30): fix: resolve timezone bug causing wrong day display in dashboard (#1032)

* fix: resolve timezone bug causing wrong day display in dashboard

Fixes issue where users in timezones ahead of UTC (e.g., UTC+12) would see
yesterday's date in the dashboard after noon local time.

Changes:
- Create shared date utility functions in utils/date.ts
- Replace toISOString() usage with local timezone helpers
- Fix DashboardPage date initialization and navigation
- Fix DailySummaryCard isToday check
- Fix DatePicker timezone handling
- Update frontend CLAUDE.md with date handling guidelines

The root cause w [... cut]