tphakala/birdnet-go #1051: New UI: Dashboard broken in latest nightly

Lenbok, 2025-07-30:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

I just updated the latest nightly, and now the new UI dashbord doesn't show the summary or recent detections.

## Steps to Reproduce (for bugs)
1. Got to the dashboard

## Screenshots

Side by side, old ui in one tab, new ui in the other:

[image]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
 nightly-20250728-15-g550d8bbf 
```

### System Information
- Device: 
  - [x] Raspberry Pi 4 Model B)

Lenbok, 2025-07-30:
I tried clearing browser cache. No messages in the browser console. I am using firefox.

There are messages if I open the web developer tools ui. I have no idea if they're important. (this is from a different firefox instance)

[image]

tphakala, 2025-07-31:
It is fixed by this PR https://github.com/tphakala/birdnet-go/pull/1053, this was a stupid error which should been detected at build time to prevent faulty release.

I am still learning about js/ts ecosystem and I had a misunderstanding that typescript code would be type checked at build time, but I now I learned that vite build system does not do that unless typecheck is exclipictily requested. I'll improve my build workflow to prevent issues like this.

Lenbok, 2025-08-01:
Yes, can confirm this is fixed.

closed 2025-08-01 by Lenbok