tphakala/birdnet-go #1069: RTSP saying unhealthy and not using

matthew73210, 2025-08-02:
## Issue Type
- [ x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
Birdnet-go is reporting that the rtsp is unhealthy, but I can connect to said stream and hear sounds perfectly clear. Maybe 10-30 seconds lag.

## Steps to Reproduce (for bugs)
1. Connect to rtsp

## Expected behavior
Work :)

## Actual behavior
Not processing data

## Screenshots
`
2025/08/02 10:14:28 🔄 Restarted FFmpeg stream for rtsp://192.168.1.33:8554
2025/08/02 10:14:28 🧹 Age-based cleanup completed successfully, clips removed: 0, current disk utilization: 75%
2025/08/02 10:14:58 ⚠️ Unhealthy stream detected: rtsp://192.168.1.33:8554 (last data: 5m31.194594208s ago)
2025/08/02 10:14:58 🔄 Restarted FFmpeg stream for rtsp://192.168.1.33:8554
2025/08/02 10:15:28 ⚠️ Unhealthy stream detected: rtsp://192.168.1.33:8554 (last data: 6m1.194540472s ago)
2025/08/02 10:15:28 🔄 Restarted FFmpeg stream for rtsp://192.168.1.33:8554
2025/08/02 10:15:58 ⚠️ Unhealthy stream detected: rtsp://192.168.1.33:8554 (last data: 6m31.194922413s ago)
2025/08/02 10:15:58 🔄 Restarted FFmpeg stream for rtsp://192.168.1.33:8554
`

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
`
    rtsp:
        transport: udp
        urls:
            - rtsp://192.168.1.33:8554/
        health:
            healthydatathreshold: 60
            monitoringinterval: 30
        ffmpegparameters: []
`

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ x] PC/Server (please specify)
- OS: HAOS
- Docker

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ x] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

Running a addon using alexbelgium repo, nightly-20250731-1-g0ff3b1fc

tphakala, 2025-08-02:
Please submit a support dump from Settings > Support and provide system ID so I can take a look at logs for more context

matthew73210, 2025-08-02:
I've upped the health constraints, i'll leave it for an hour and get back to you. Cheers

closed 2025-08-07 by tphakala

matthew73210, 2025-08-07:
Ah i get it ! No it's not unstable, the rtsp server has to restard every so often (cron). A feature if you will. This might be causing birdnet to think that it's unstable

PR #1106 (2025-08-07): fix: resolve RTSP stream infinite restart loop
## Summary
Fixes #1069 - RTSP stream marked as unhealthy and continuously restarting

This PR resolves the circuit breaker reset timing bug that caused infinite restart loops for unreachable RTSP sources. The core issue was that failures were being reset prematurely on process start, before the stream had proven stable operation.

## Problem
- Circuit breaker failures were reset immediately when FFmpeg process started
- This allowed infinite restart loops for unreachable RTSP sources
- Streams appeared "healthy" before receiving any data
- Circuit [... cut]
commit 37237aae (2025-08-09): fix: resolve RTSP stream infinite restart loop (#1106)

* fix: resolve RTSP stream infinite restart loop (fixes #1069)

This commit fixes the circuit breaker reset timing bug that caused infinite
restart loops for unreachable RTSP sources.

## Key Changes

### Circuit Breaker Fix
- Remove premature failure reset on process start
- Add conditional reset only after stable operation (30s runtime + 100KB data)
- Implement graduated failure thresholds based on runtime:
  - < 1 second: Opens after 3 failures (immediate connection failures)
  - < 5 seconds: Opens after 5 failures (rapid process failu [... cut]