tphakala/birdnet-go #1179: Install Failure on Pi 4 B

stevedee, 2025-08-20:
Support please, new to BirdNet-Go.

Tried to install twice on a Raspberry Pi 4, but after entering initial settings, system will not run.

Here are the log entries:-
🚀 Starting BirdNET-Go...
✅ BirdNET-Go service started successfully!

🐳 Waiting for container to start...
⏳ Waiting for container to start (attempt 1/30)...
❌ Service stopped unexpectedly
Checking service logs:
Aug 20 09:42:15 birdnet-test docker[999]: Error: No such container: birdnet-go
Aug 20 09:42:15 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.
Aug 20 09:42:16 birdnet-test docker[1014]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (dd2a39b591027b428fda42e85f4e17d9d3f4cd172ca823c6d74a6240c4185c33): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
Aug 20 09:42:16 birdnet-test systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Aug 20 09:42:16 birdnet-test docker[1133]: Error: No such container: birdnet-go
Aug 20 09:42:16 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 09:42:17 birdnet-test systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 2.
Aug 20 09:42:17 birdnet-test systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Aug 20 09:42:17 birdnet-test systemd[1]: Starting birdnet-go.service - BirdNET-Go...
Aug 20 09:42:17 birdnet-test docker[1141]: Error: No such container: birdnet-go
Aug 20 09:42:17 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.
Aug 20 09:42:18 birdnet-test docker[1156]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (177d7587366f3461d58fb36c2b89da4f0f9aadcd58523c767ef0077005eb72a5): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
Aug 20 09:42:18 birdnet-test systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Aug 20 09:42:18 birdnet-test docker[1274]: Error: No such container: birdnet-go
Aug 20 09:42:18 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 09:42:18 birdnet-test systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 3.
Aug 20 09:42:18 birdnet-test systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Aug 20 09:42:18 birdnet-test systemd[1]: Starting birdnet-go.service - BirdNET-Go...
Aug 20 09:42:18 birdnet-test docker[1282]: Error: No such container: birdnet-go
Aug 20 09:42:18 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.
Aug 20 09:42:19 birdnet-test docker[1297]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (7d91dee42be83eee2404d7e1ed23f3561fb771dc2e08430dd4e6ca1000bfee7a): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
Aug 20 09:42:19 birdnet-test systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Aug 20 09:42:19 birdnet-test docker[1412]: Error: No such container: birdnet-go
Aug 20 09:42:19 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 09:42:20 birdnet-test systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 4.
Aug 20 09:42:20 birdnet-test systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Aug 20 09:42:20 birdnet-test systemd[1]: Starting birdnet-go.service - BirdNET-Go...
Aug 20 09:42:20 birdnet-test docker[1419]: Error: No such container: birdnet-go
Aug 20 09:42:20 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.
Aug 20 09:42:21 birdnet-test docker[1434]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (379a2fe5b8934b8a1da9da39d049316dbe1212c72e0117afa4962d6834b31410): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
Aug 20 09:42:21 birdnet-test systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Aug 20 09:42:21 birdnet-test docker[1549]: Error: No such container: birdnet-go
Aug 20 09:42:21 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 09:42:21 birdnet-test systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 5.
Aug 20 09:42:21 birdnet-test systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Aug 20 09:42:21 birdnet-test systemd[1]: birdnet-go.service: Start request repeated too quickly.
Aug 20 09:42:21 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 09:42:21 birdnet-test systemd[1]: Failed to start birdnet-go.service - BirdNET-Go.
Aug 20 16:44:08 birdnet-test systemd[1]: Starting birdnet-go.service - BirdNET-Go...
Aug 20 16:44:08 birdnet-test docker[2771]: Error: No such container: birdnet-go
Aug 20 16:44:08 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.
Aug 20 16:44:09 birdnet-test docker[2785]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (d8d85dc6ab1ad9f89e52e999f02ab70eefd97f875b674c6c21c7ffd8c28d77ef): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
Aug 20 16:44:09 birdnet-test systemd[1]: birdnet-go.service: Main process exited, code=exited, status=125/n/a
Aug 20 16:44:09 birdnet-test docker[2955]: Error: No such container: birdnet-go
Aug 20 16:44:09 birdnet-test systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
Aug 20 16:44:09 birdnet-test systemd[1]: birdnet-go.service: Scheduled restart job, restart counter is at 1.
Aug 20 16:44:09 birdnet-test systemd[1]: Stopped birdnet-go.service - BirdNET-Go.
Aug 20 16:44:09 birdnet-test systemd[1]: Starting birdnet-go.service - BirdNET-Go...
Aug 20 16:44:09 birdnet-test docker[2969]: Error: No such container: birdnet-go
Aug 20 16:44:09 birdnet-test systemd[1]: Started birdnet-go.service - BirdNET-Go.

❗ If you need help with this issue:
1. The service started but then crashed

---end of log----------

Provide the following line from your system:-

$ ./birdnet-go
-bash: ./birdnet-go: No such file or directory

### System Information

- Raspberry Pi 4 Model B
  - RAM: 2GB
  - Storage type and size:  32GB microSD card
  - microphone audio from RTSP
  - 
steve@birdnet-test:~$ uname -a
Linux birdnet-test 6.12.34+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

tphakala, 2025-08-20:
Is this a fresh install of Raspberry Pi OS?
Is there anything else running on this Raspberry Pi OS?

This indicates that something else is on port 80, but this is also add as default port of BirdNET-Go is 8080

`Aug 20 09:42:18 birdnet-test docker[1156]: docker: Error response from daemon: driver failed programming external connectivity on endpoint birdnet-go (177d7587366f3461d58fb36c2b89da4f0f9aadcd58523c767ef0077005eb72a5): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.
`

Did you install with these exact commands?

```
curl -fsSL https://github.com/tphakala/birdnet-go/raw/main/install.sh -o install.sh
bash ./install.sh
```

stevedee, 2025-08-20:
OK, I'm an idiot.

With a clean OS, BirdNet-Go installed no probs.

Thanks once again tphakala.

I'm running this alongside Nachtzuster's BirdNet-Pi using a common mic, so looking forward to comparing results over the coming weeks!

tphakala, 2025-08-20:
Good that you were able to sort it out! Please test new UI and provide feedback, see this https://github.com/tphakala/birdnet-go/discussions/1097 for more details.