tphakala/birdnet-go #1184: Lamb Call gets confused for Hadada Ibis

HinzundKunz, 2025-08-21:
That's pretty much it. I put in a filter for Hadada Ibis to only record it if it's 100% sure, and added Hadada Ibis to the dog filter, but I don't know whether that was the right thing to do, as lambs and dogs don't count as the same species... ;-)

tphakala, 2025-08-21:
This is BirdNET AI model issue, nothing much I can do about it in application level.

tphakala, 2025-08-21:
No it does not solve this, dog filter works by tracking dog bark detections, if there is multiple dog barks and sudden bird detection that bird detection will be ignored, like:

dog bark
dog bark
dog bark
dog bark
eagle owl <- dog bark mis detected as owl
dog bark
dog bark
dog bark

that eagle owl detection would be ignored it eagle owl was added to dog bark filter list. 
Unless lamb is detected as dog bark it does not work here.

However, if you could record those lamb sounds it would be easy to add lamb detection by training a custom classifier which would then remove false detection.

HinzundKunz, 2025-08-21:
I've uploaded the files here:
https://drive.google.com/drive/folders/16X43Y1Xv1wgwGyzBlCItL-1wSCQPgIZb?usp=sharing

HinzundKunz, 2025-08-22:
I have a couple more that have been misidentified as Trumpeter Hornbill if you want them... ;-)

HinzundKunz, 2025-08-22:
As they are, or normalized?

HinzundKunz, 2025-08-22:
Ok, again the same folder, but a subfolder:
https://drive.google.com/drive/folders/16X43Y1Xv1wgwGyzBlCItL-1wSCQPgIZb?usp=sharing

ModerateWinGuy, 2025-08-24:
> Yes, please. Training a custom classifier requires hundreds of audio samples of species for good performing model.

What if we added a feature where you could define a custom server upload address via ftp or something and every call that you reviewed + verified got uploaded to that storage so we could easily contribute to something like this for you?

ModerateWinGuy, 2025-08-24:
That's fair enough, and I suppose the possible future feature I saw you mention somewhere, to export all correctly reviewed detections, would get people most of the way there to export a bunch of clips and send them to someone who wants them.

HinzundKunz, 2025-08-25:
BTW, I have a box.com account of 50GB that I never use... Would that be of any help?

tphakala, 2025-08-25:
If you are asking from me, classifier training based on user submitted recordings is not an IT resource problem, it is a human resource problem. To keep classifier reliable (avoid training it with incorrect data) it cannot be automated very easily, every recording has to be verified, and often also edited manually.

ModerateWinGuy, 2025-08-28:
I don't suppose you have some time to write instructions on how you trained it?
I'd love to add some more fine tuning for my local birds; I downloaded the birdnet-analyzer app and ran into a bunch of weird errors when using the train option.

tphakala, 2025-08-30:
> I don't suppose you have some time to write instructions on how you trained it? I'd love to add some more fine tuning for my local birds; I downloaded the birdnet-analyzer app and ran into a bunch of weird errors when using the train option.

I did manage to do documentation for this, not necessarily 100% complete yet but I hope this gets you going https://github.com/tphakala/birdnet-go/blob/main/doc/wiki/training-a-custom-classifier.md

HinzundKunz, 2025-09-01:
Ok, Thanks a lot. More (stupid ;-) ) follow-up questions:
1) If I create a folder structure like you laid out in the wiki on my box.com account, do you think the birdnet team would be able to use it for their training?
2) Today I had an Eastern Violet-backed Sunbird that was misidentified as a Eurasian blackcap. Should I 
    a) put the recordings in a subfolder for EVbS, and
    b) additionaly put them into a "-sylvia atricapilla_Eurasian Blackcap" folder?