tphakala/birdnet-go #1223: Update to nightly blank dashboard

nommo-san, 2025-08-26:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type

- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

Nightly ./install.sh update resulted in blank dashboard, on investigation:

*Location data was reset, I manually added lat/long same as my Birdweather station.
*Birdweather integration was reset (I readded)
*Audio settings were reset and the RTSP url was gone (I readded)
*The dog bark filter was checked after the update, I never checked it before.

There are probably more things I am not noticing.

I am now able to see new detections in the UI but no historical detections in the dashboard or analytics. Getting a lot of new species detected notifications now.

## Steps to Reproduce (for bugs)

1. sudo bash ./install.sh

## Expected behavior

Update with bug fixes

## Actual behavior
 
Like a fresh install. Database still populated but Birdnet-go UI not pulling in data

## Screenshots

[If applicable, add screenshots to help explain your problem]

## Environment

### Version Information (build date)

Provide the following line from your system:

```
$ ./birdnet-go
pi@pi5:~ $ ./birdnet-go
-bash: ./birdnet-go: No such file or directory

```

### System Information

- Device:
  - [x] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: Raspberry Pi OS
- OS Version: Raspberry Pi OS Bookworm
- Hardware specs:
  - CPU: Raspberry Pi 5 Broadcom BCM2712 64-bit Arm Cortex-A76 
  - RAM: 8GB
  - Storage type and size: 256GB M.2 SSD

tphakala, 2025-08-26:
Yeah I think it is a browser cache issue, @nommo-san try doing a browser cache clear and hard reload by opening browser developer  mode with F12 and right clicking reload. I added some cache busting features in latest nightly release which should prevent this from happening, but I just experienced some issues myself so it might not be perfect yet.

nommo-san, 2025-08-27:
Hi @tphakala thanks for looking into this. Yesterday, I did try clearing cache on Firefox, tried using Incognito, switched to Chrome on desktop PC and android mobile but that made no difference. So I figured it must be something else and posted here.

Following your guidance, in Chrome just now, I deleted the cookies and data for my install domain, F12, right click, empty cache and hard reload and I am still getting no detections prior to Tuesday when I readded the RTSP stream etc.

I may need to stick to release... might be for the best for someone who needs to google 'docker rebase' (and is still none the wiser!).

tphakala, 2025-08-27:
Could your issue be related to this? https://github.com/tphakala/birdnet-go/issues/1221
Issue described in that issue was fixed in nightly release done yesterday

nommo-san, 2025-08-27:
Hmm maybe @tphakala - I didn't get the error notifications though... and although I am using a Reolink security camera like @Lenbok I do not have special characters in my password. I'll try an update to the latest nightly and hope that fixes it!

nommo-san, 2025-08-27:
OK, that seems to have sorted it whatever it was!

closed 2025-08-30 by tphakala