tphakala/birdnet-go #1234: Cannot login in when using /ui endpoint

sve007, 2025-08-29:
Please fill in this issue template, I cannot help if I don't have basic details like application version etc.

## Issue Type

- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

When I use the web interface through the /ui endpoint, logging in with my configured password will not work, giving the following error message:
**Invalid credentials. Please try again.**

Navigating to the standard UI and logging in with the same password works every time.

## Steps to Reproduce (for bugs)

1. Go to /ui
2. Click the Log in button
3. Enter correct password

## Expected behavior

The use should be logged in.

## Actual behavior

Password is rejected

## Screenshots

## Environment

### Version Information (build date)

Provide the following line from your system:

```
BirdNET-Go nightly-20250826-9-g27721729 (built: 2025-08-26T16:57:40Z), using config file: /root/.config/birdnet-go/config.yaml
```

### System Information

- Device:
  - [ ] Raspberry Pi 4B
- 
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: Raspberry Pi OS
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: 4GB
  - Storage type and size: 32GB microSD card
  - 
### For Non-Raspberry Pi Linux Users

Have you tested with the latest dev branch container image?

- [ ] Yes
- [ ] No
- [ x] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.

```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
# Alternative: docker pull tphakala/birdnet-go:dev
```

## Additional context

[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution

[If you have a suggestion for fixing the bug or implementing the feature]

sve007, 2025-08-29:
Good to know a fix is in the works. Unfortunately reconfiguring through the old UI does not seem te fix the issue. But since a fix is coming, the workaround by switching ui's is fine.

jwtobias, 2025-11-04:
Just ran into this as well. It seems that the new UI may also hide the option to log in with Google OAuth 2.0 as well.

nicopasla, 2025-12-15:
This still happen to me except it happen every time i'm using the cloudflare tunnel, the local adress work and show OAuth and basic password work but on cloudflare tunnel i get 'Invalid credentials. Please try again.'