tphakala/birdnet-go #1245: rpi zero 2w: ringbuffer full

philou155, 2025-08-31:
trying to use birdnet-go on a Raspberry Pi Zero 2 (512MB).

I disabled deep detection and dynamic threshold... are there features i should disable or not to make it work? it seems to still be unstable and i get an error message:  "ringbuffer full"

system ID 42C3-106E-9300. I submited a support report earlier.

tphakala, 2025-08-31:
Ok sorry, problem is that I don't see config file as it is on your system. Support dump sanitizes config quite hard removing private info, problem is that it is a bit too aggressive so it also replaces empty values like rtsp source '' with rtsp source '[REDACTED]' so to me it looks that there might be RTSP URL configured.

I need to improve this.

tphakala, 2025-08-31:
> system-resource error in support: no journald logs available

Aha! This explain why your support dump did not contain any logs, it failed on journald collection and interrupted whole collection process. This is now fixed with https://github.com/tphakala/birdnet-go/pull/1247. I also doubled size of ring buffer size in hopes of reducing errors you are seeing.

Could you try to install latest dev container image to test these fixes? (I assume you are using container based install). I updated install.sh to accept --version parameter so if you download fresh copy of install.sh and run

`bash ./install.sh --version dev`

You can switch back to nightly releases with

`bash ./install.sh --version nightly`

Which installs latest nightly snapshot version.

But in short, this is some resource contention issue on your system, BirdNET analysis process is not processing audio samples quickly enough from analysis ring buffer causing it to fill up.

philou155, 2025-08-31:
ok, I installed the dev version and looks like i don't have the ringbuffer issue anymore (so far)

I tried to generate a support report again but got the same error:
system-resource error in support: no journald logs available

i did get this message first:
Support dump successfully uploaded to developers! Reference ID: 4801ee69-5f52-4205-9195-7b71b81872de

thanks again