tphakala/birdnet-go #1249: RTSP URLs break if they contain an ampersand

skullydazed, 2025-08-31:
## Issue Type

- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

I have some amcrest cameras that use this RTSP url- rtsp://USER:PASS@192.168.1.100:554/cam/realmonitor?channel=1&subtype=0

When I setup birdnet-go using the script I pasted the URL into the setup script, which resulted in a mangled URL and log messages like this:

```
Aug 31 14:03:53 lowpass docker[101612]: 2025/08/31 14:03:53 ⚠ Warning: failed to remove analysis buffer for rtsp://192.168.1.100:554/cam/realmonitor?channel=1#%20-%20rtsp://user:password@example.com/stream1subtype=0: no ring buffer found for source ID: rtsp://USER:PASS@192.168.1.100:554/cam/realmonitor?channel=1# - rtsp://user:password@example.com/stream1subtype=0
Aug 31 14:03:53 lowpass docker[101612]: 2025/08/31 14:03:53 ⚠ Warning: failed to remove capture buffer for rtsp://192.168.1.100:554/cam/realmonitor?channel=1#%20-%20rtsp://user:password@example.com/stream1subtype=0: no capture buffer found for source: rtsp://USER:FUCKYOU@192.168.1.100:554/cam/realmonitor?channel=1# - rtsp://user:password@example.com/stream1subtype=0
Aug 31 14:03:53 lowpass docker[101612]: 2025/08/31 14:03:53 🛑 Stopped FFmpeg stream for rtsp://192.168.1.100:554/cam/realmonitor?channel=1#%20-%20rtsp://user:password@example.com/stream1subtype=0
```

Going to Settings -> Audio Capture -> RTSP Stream URLs and correcting the URL, whether by editing the existing one or removing and adding new, results in a new error:

```
Aug 31 14:06:54 lowpass docker[102454]: 2025/08/31 14:06:54 ❌ Failed to register RTSP source rtsp://192.168.1.100:554/cam/realmonitor?channel=1&subtype=1: dangerous pattern detected in connection string
Aug 31 14:06:54 lowpass docker[102454]: 2025/08/31 14:06:54 ⚠  1 out of 1 RTSP sources failed to register: [rtsp://192.168.1.100:554/cam/realmonitor?channel=1&subtype=1]
Aug 31 14:06:54 lowpass docker[102454]: 2025/08/31 14:06:54 ⚠  Error initializing buffers: buffer initialization errors: failed to initialize analysis buffers: no audio sources provided; failed to initialize capture buffers: no capture sources provided
Aug 31 14:06:54 lowpass docker[102454]: 2025/08/31 14:06:54 ⚠  Some audio sources might not be available.
```

If I remove the `&subtype=1` from the URL I don't get any errors.

## Steps to Reproduce (for bugs)

1. Try to use RTSP that requires an & in the URL

## Expected behavior

Audio working

## Actual behavior

Audio not working

## Screenshots

[If applicable, add screenshots to help explain your problem]

## Environment

### Version Information (build date)

Provide the following line from your system:

```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

How do I do this after using https://github.com/tphakala/birdnet-go/raw/main/install.sh?

### System Information

- Device:
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [x] PC/Server (please specify)
- OS: Debian
- OS Version: Bookworm
- Hardware specs:
  - CPU: intel i77-8550U
  - RAM: 16gb
  - Storage type and size: 500gb NVME
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: Amcrest 5MP camera 

### For Non-Raspberry Pi Linux Users

Have you tested with the latest dev branch container image?

- [x] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.

```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
# Alternative: docker pull tphakala/birdnet-go:dev
```

## Additional context

This is a brand-new setup using the install.sh script on a brand-new debian install, nothing else was installed to this machine.

## Possible Solution

[If you have a suggestion for fixing the bug or implementing the feature]

skullydazed, 2025-09-01:
@tphakala thanks for the fast turnaround here, unfortunately it looks like the nightly docker build failed, I'm still getting the same behavior even after stopping birdnet-go, running `docker system prune -a`, and reinstalling.

tphakala, 2025-09-01:
Install dev image until nightly build is available, download fresh copy of install.sh and run it with --version dev parameter

commit e7001f3f (2025-10-01): fix: allow ampersands in RTSP URL query parameters (#1257)

Resolves issue where RTSP URLs with query parameters containing
ampersands were incorrectly flagged as dangerous patterns. RTSP URLs
like "rtsp://user:pass@ip:port/path?channel=1&subtype=0" are now
properly validated.

Changes:
- Skip dangerous pattern validation for RTSP URLs since they have
  their own specific validation
- RTSP URLs are passed directly to FFmpeg which handles them safely
- Add comprehensive test cases for RTSP URLs with query parameters
- Maintain strict security checks for file paths and other sources

Fixes #1249 [... cut]