tphakala/birdnet-go #1264: Audio detection via RTSP hangs after less than a day

seajseven, 2025-09-03:
Set up a Birdnet-go instance on a RaspberryPi 5 in Docker and let it run for a few weeks. (Audio input is via RTSP from an Ethernet Unifi camera.) After some time running, the Audio detection stops ( indicator at top of webpage stops animating ) and ffmpeg does not seem to be running. Once the issue starts, it seems to require a reboot, and then it will return after a day or so. 

System ID: 58B6-BBB9-1FCC

tphakala, 2025-09-04:
By looking at logs RTSP source has stopped sending data, BirdNET-Go restarts and connects to stream correctly but source never sends anything back. Logs from few people look similar, I think there is something special with Unifi cameras, but I can't test and debug it as I don't have one.

tphakala, 2025-09-20:
@brujoand can you please send support report through Settings > Support?

brujoand, 2025-09-20:
> [@brujoand](https://github.com/brujoand) can you please send support report through Settings > Support?

Sure thing:
```
Support dump successfully uploaded to developers! Reference ID: f1c2525b-9b77-4393-9e99-4334247b353f
```

lbatalha, 2025-10-17:
I have been running that nightly for the last ~3 days and it appears to successfully recover from hung streams, the new status API endpoints are also extremely handy to figure out what is happening.
If I spot any permanent hangs, I'll report back.