tphakala/birdnet-go #1316: Season hemisphere detection

ModerateWinGuy, 2025-09-22:
## Issue Type

- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description

The "First this [season]" dialogue isnt taking location in to account.
It is spring here in the southern hemisphere now, not fall.

[image]

## Steps to Reproduce (for bugs)

1. Detect a species for the first time that season
## Expected behavior

[What you expected to happen]

##  It should say Fall or Spring depending on the hemisphere location of the station

[What actually happened]

## Screenshots

[If applicable, add screenshots to help explain your problem]

## Environment

### Version Information (build date)
nightly-20250825-1-g3b4f4c08

Provide the following line from your system:

```bash
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

Or copy paste release information from Web UI

### System Information

- Device:
  - [ Docker ] PC/Server (please specify)
- OS: [TrueNas]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users

Have you tested with the latest dev branch container image?

- [ x ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.

```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
# Alternative: docker pull tphakala/birdnet-go:dev
```

## Additional context

[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution

When setting the station location, set a northern / southern hemisphere flag depending on the location.

ModerateWinGuy, 2025-09-22:
Alright, i can take a look at it and get a pr up if you haven't got one in progress already.

tphakala, 2025-10-08:
This is fixed in latest nightly

closed 2025-10-08 by tphakala

StrontiumSteel, 2025-12-04:
> It seems this is not fully fixed, out there is another bug - since the first of December all my detections started showing the leaf symbol, with tooltip saying first time this fall (instead of summer).

Yep - same here. I came here to say exactly that.